arnemauer/Ducobox-ESPEasy-Plugin #113: Kan geen verbinding maken met Enri CC1101 shield

richardvenneman, 2022-08-15:
Hoi! Ik ben erg enthousiast om de Ducobox bij ons thuis aan te sturen en vond een kant-en-klare oplossing van enri.nl: https://enri.nl/pcb/?page=itho.

Ik heb de nieuwste versie van de firmware in dit repository geflashed. De instellingen heb ik overgenomen van eerdergenoemde URL: 

[image]

Het klikken van Join levert wel wat log-informatie op, maar het lijkt erop dat er nooit een antwoord komt van de Ducobox:

```
4228: Current Time Zone: STD time start: 2022-10-30 03:00:00 offset: 0 min
4230: Local time: 2022-08-15 21:35:06
31750: WD : Uptime 1 ConnectFailures 0 FreeMem 18768 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
60496: [P150] RF GW: UPDATE by PLUGIN_READ
61750: WD : Uptime 1 ConnectFailures 0 FreeMem 18016 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
73412: HTTP: JOIN
73415: Command: JOIN
73416: [P150] RF GW: Sent command for 'join' to DUCO unit
73423: [P150] RF GW: 0 - SendJoinPacket()
73423: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
73423: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
73423: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
73423: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
91750: WD : Uptime 2 ConnectFailures 0 FreeMem 18464 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
[...]
```

De Ducobox staat in pairing mode. Raspberry Pi voeding en 3 meter afstand tot de Ducobox. Mis ik iets? :)

richardvenneman, 2022-08-23:
[image]

arnemauer, 2022-08-23:
De module en antenne zien er prima uit. 
De initialisatie van de cc1101 lijkt goed te gaan. Zie je wel berichten in de weblog voorbij komen als je wat knoppen indrukt op een duco afstandsbediening?

richardvenneman, 2022-08-23:
> Zie je wel berichten in de weblog voorbij komen als je wat knoppen indrukt op een duco afstandsbediening?

Nee helaas, ik zie geen berichten in de weblog voorbij komen anders dan de uptime berichten. Ik heb het zojuist nog een aantal keer geprobeerd.

aiolos, 2022-08-24:
Maker van de shield hier :)
Nu heb ik zelf geen Duco, dus daar kan ik niet mee testen, maar initialisatie en antenne waren voor versturen goed, want op mijn Itho werkte het wel (daar test ik ze altijd op).

Kan me alleen herinneren dat ik sinds kort de nieuwste binary ben gaan flashen, waar ik daarvoor een oudere versie had. Kan er in die nieuwste versie iets mis zijn? Een oudere binary flashen kan dan wellicht helpen.

BSW90, 2022-08-24:
Hoi, hier iemand met hetzelfde probleem. Ik gebruik dezelfde oplossing als jij, tevens in combinatie met de Ducobox silent. De box staat in installer mode, op het moment dat ik probeer te pairen komt de melding van het request wel voorbij, maar daarbij houdt het op; geen response van de Duco box;

```
04246: HTTP: JOIN
704250: [P150] RF GW: Sent command for 'join' to DUCO unit
704257: [P150] RF GW: 704250 - SendJoinPacket()
704258: [P150] RF GW: 704256 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
721839: WD   : Uptime 12 ConnectFailures 0 FreeMem 14328 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
751839: WD   : Uptime 13 ConnectFailures 0 FreeMem 13624 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
754000: [P150] RF GW: Extra Settings PLUGIN_150 loaded
754012: [P150] RF GW: Values set from config. DeviceID: 0, networkId: 0000, radio power: Default - radio power 0,6 dBm (0x8D)
754023: [P150] RF GW: CC1101 868Mhz transmitter initialized
754133: [P150] RF GW: 754015 - sendSubscribeMessage();
754133: [P150] RF GW: 754016 - SENT message (outboxQ 0): SRC:0; DEST:1; ORG.SRC:0; ORG.DEST:1; Ntwrk:00000000;Type: 7;
754134: [P150] RF GW: 754016 - Bytes:18; Counter:3; WaitForAck: 1;
754135: [P150] RF GW: 754016 - 00,07,00,00,00,00,00,40,13,00,60,00,22,00,00,6E,FF,00,
754136: [P150] RF GW: 754023 - Start waiting for ack...
781839: WD   : Uptime 13 ConnectFailures 0 FreeMem 13936 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
811840: WD   : Uptime 14 ConnectFailures 0 FreeMem 14424 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
841839: WD   : Uptime 14 ConnectFailures 0 FreeMem 14664 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
871840: WD   : Uptime 15 ConnectFailures 0 FreeMem 14416 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
901840: WD   : Uptime 15 ConnectFailures 0 FreeMem 14088 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
931839: WD   : Uptime 16 ConnectFailures 0 FreeMem 14320 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
961839: WD   : Uptime 16 ConnectFailures 0 FreeMem 14216 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

Ik ben al teruggegaan van ESP_Easy_mega_20220416_custom_ESP8266_4M1M naar ESP_Easy_mega_20200629_normal_ESP8266_4M1M, maar dit maakt geen verschil.

arnemauer, 2022-08-24:
De rf frequentie is een aantal versies geleden veranderd. Ik zal vanavond even checken welke versie dit is geweest. Kunnen jullie aangeven welke ducobox jullie hebben en welke hardwareversie en bouwjaar hij heeft?

richardvenneman, 2022-08-24:
Ik heb ook verschillende firmware versies geprobeerd (tot aan 2020 als ik me niet vergis), maar zonder resultaat.

Ik weet niet precies wat het bouwjaar is van deze Ducobox Silent, maar woon ik een appartement wat ongeveer 4 jaar geleden is opgeleverd.

[image]

BSW90, 2022-08-24:
Hier is de box in 2016 geplaatst. Artikelnummer lijkt hetzelfde te zijn, versie van het printje zo te zien iets ouder. 

[image]
[image]

aiolos, 2022-08-24:
Ik heb het nog even teruggezocht en eerder flashte ik `20200608 Firmware aa380ce 08-06-2020 (ESPEasy mega-20200515).bin` 
Dat is dus al best een oude versie...

richardvenneman, 2022-08-25:
> Ik heb het nog even teruggezocht en eerder flashte ik `20200608 Firmware aa380ce 08-06-2020 (ESPEasy mega-20200515).bin`
> Dat is dus al best een oude versie...

Ik heb zojuist versie `firmware 17-04-2020 5177743.bin` ([link](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/6c6b4d78cac08502fe48093e89d8b139fb7a9ffc/Binaries/Binary%20based%20on%20ESPEasy_mega-20200222/firmware%2017-04-2020%205177743.bin)) geflashed.

```
Firmware
Build:⋄ | 20104 - Mega
System Libraries:⋄ | ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | AFM 5177743: Apr 17 2020
Plugin Count:⋄ | 54 [Normal]
Build Time:⋄ | Apr 17 2020 13:54:41
Binary Filename:⋄ | Self built!

```

.. en het proces nogmaals doorlopen. Helaas zonder resultaat:

<details><summary>Log</summary>

```
420947: [P150] DUCO RF GW: Values set from config. DeviceID: 0 and networkId: 0000
420949: [P150] DUCO RF GW: CC1101 868Mhz transmitter initialized
421050: [P150] DUCO RF GW: TIMER => 421050 420947 103
421050: [P150] DUCO RF GW: Start of RF signal received
421051: [P150] DUCO RF GW: Ignoring RF noise
421057: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
421058: [P150] DUCO RF GW: State: 0.00
422027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
422027: [P150] DUCO RF GW: State: 0.00
423027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
423028: [P150] DUCO RF GW: State: 0.00
423895: WD : Uptime 7 ConnectFailures 0 FreeMem 12184 WiFiStatus 3
424027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
424027: [P150] DUCO RF GW: State: 0.00
425027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
425027: [P150] DUCO RF GW: State: 0.00
426027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
426027: [P150] DUCO RF GW: State: 0.00
427027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
427027: [P150] DUCO RF GW: State: 0.00
428027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
428027: [P150] DUCO RF GW: State: 0.00
429027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
429027: [P150] DUCO RF GW: State: 0.00
430027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
430027: [P150] DUCO RF GW: State: 0.00
431027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
431027: [P150] DUCO RF GW: State: 0.00
432027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
432027: [P150] DUCO RF GW: State: 0.00
433027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
433027: [P150] DUCO RF GW: State: 0.00
434027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
434027: [P150] DUCO RF GW: State: 0.00
435027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
435027: [P150] DUCO RF GW: State: 0.00
436027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
436027: [P150] DUCO RF GW: State: 0.00
437027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
437027: [P150] DUCO RF GW: State: 0.00
438027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
438027: [P150] DUCO RF GW: State: 0.00
439027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
439027: [P150] DUCO RF GW: State: 0.00
440027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
440027: [P150] DUCO RF GW: State: 0.00
441027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
441027: [P150] DUCO RF GW: State: 0.00
442027: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
[... 79 lines cut ...]
478553: [P150] DUCO RF GW: 478552 - Joinpacket sent
478627: [P150] DUCO RF GW: TIMER => 478627 478551 76
478628: [P150] DUCO RF GW: Start of RF signal received
478628: [P150] DUCO RF GW: Ignoring RF noise
479028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
479028: [P150] DUCO RF GW: State: 0.00
480028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
480028: [P150] DUCO RF GW: State: 0.00
481028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
481028: [P150] DUCO RF GW: State: 0.00
482028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
482028: [P150] DUCO RF GW: State: 0.00
483028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
483028: [P150] DUCO RF GW: State: 0.00
483895: WD : Uptime 8 ConnectFailures 0 FreeMem 11896 WiFiStatus 3
484028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
484028: [P150] DUCO RF GW: State: 0.00
485028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
485028: [P150] DUCO RF GW: State: 0.00
486028: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
486029: [P150] DUCO RF GW: State: 0.00
```
</details>

Ik zie dat er meerdere versies gebaseerd zijn op die build, dus misschien kan ik een oudere versie proberen..

arnemauer, 2022-08-26:
De verandering in de frequentie is op 29 mei 2022 gebeurt. Dus dat kan het probleem niet zijn.
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/062a780e71de3003a5c42453f098d9927d105ca7#diff-873fd31b78f810ad0ff22a54159143e2adb86cf6dc98b5a6986564e1cb08762aL129

In de logs zie ik geen interrupt voorbij komen dat er een pakketje ontvangen is. 
Wat zie je in de logs wanneer je het commando "CC1101STATUS" geeft?
En wat bij het commando "RXBYTES"?

Jover68, 2022-08-27:
[image]
Ik heb de onze gisteren een update gegeven. Hij kwam van firmware 15-03-2020 bac5c26.bin en is naar de laatste versie gegaan. Voorheen werkte hij wel, na de update werkt de RF niet meer en krijg ik in de log de melding "not our network ID". Is dit makkelijk te verhelpen?

Jover68, 2022-08-27:
[image]
Hij werkt weer. Bij Network ID (HEX): stond 00000000 en mag je niet handmatig invoeren. Bij het opvragen van de CC1101STATUS het Ntwrk-ID gevonden en ingevoerd bij Network ID (HEX):. Nu werkt hij weer, ook vanuit Home Assistant.

richardvenneman, 2022-08-27:
> De verandering in de frequentie is op 29 mei 2022 gebeurt. Dus dat kan het probleem niet zijn. [062a780#diff-873fd31b78f810ad0ff22a54159143e2adb86cf6dc98b5a6986564e1cb08762aL129](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/062a780e71de3003a5c42453f098d9927d105ca7#diff-873fd31b78f810ad0ff22a54159143e2adb86cf6dc98b5a6986564e1cb08762aL129)
> 
> In de logs zie ik geen interrupt voorbij komen dat er een pakketje ontvangen is. Wat zie je in de logs wanneer je het commando "CC1101STATUS" geeft? En wat bij het commando "RXBYTES"?

Bedankt voor de reactie. Ik heb zojuist de nieuwste firmware versie teruggezet (`20220704 Firmware 417d280 04-07-2022 (ESPEasy mega-20220427).bin`).

De output van de commando's:

```
CC1101STATUS => OK
110822: HTTP: CC1101STATUS
110826: Command: CC1101STATUS
110828: [P150] RF GW: getMarcState: 0D

RXBYTES => OK
141743: HTTP: RXBYTES
141746: Command: RXBYTES
141748: [P150] RF GW: rxBYTES: 00
```

arnemauer, 2022-08-27:
Als je het commando RX bytes een aantal keer achter elkaar geeft en ondertussen druk je elke keer op een knop van de afstandsbediening zie je dan wel andere waardes?

Volgende test:
Je stelt handmatig het network id van jou ducobox in en deviceid 20. Daarna geef  je het volgende commando: "DUCOTESTMESSAGE,7,2,1,2,1,600022000E6E000B"

Dit stuurt een bericht naar de ducobox (afkomstig van node 2, ik ga er vanuit dat die bestaat) om de ventilatiestand te wijzigen. Reageert de ducobox daar op door harder te gaan draaien?

richardvenneman, 2022-08-28:
Ik zie helaas geen andere waardes, enkel:
```
279422: HTTP: RXBYTES
279426: Command: RXBYTES
279427: [P150] RF GW: rxBYTES: 00
```

Voor de duidelijkheid: ik heb het commando meerdere keren uitgevoerd en drukte op willekeurige knoppen op een van de afstandsbediengen (CO2 Ruimtesensor (RF + Wired)).

Voor het handmatig instellen van de Ducobox en en DeviceID: Ik weet het Network ID van mijn ducobox niet; hoe kan ik daar achter komen? Bij het invoeren van de waardes die je hierboven geeft krijg ik het volgende:
```
447587: [P150] RF GW: EXIT PLUGIN_150
447775: [P150] RF GW: Values set from config. DeviceID: 2, networkId: 600220, radio power: Default - radio power 0,6 dBm (0x8D)
447778: [P150] RF GW: CC1101 868Mhz transmitter initialized
447786: [P150] RF GW: 0 - sendSubscribeMessage();
447786: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
447786: [P150] RF GW: 2 - Bytes:18; Counter:2; WaitForAck: 1;
447786: [P150] RF GW: 3 - 00,07,60,00,22,00,10,44,12,00,60,00,22,00,00,6E,FF,00,
447786: [P150] RF GW: 4 - Start waiting for ack...
447823: [P150] RF GW: PLUGIN_WEBFORM_LOAD
447932: [P150] RF GW: UPDATE by PLUGIN_READ
448111: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
448111: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
448111: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
448111: [P150] RF GW: 3 - Bytes:18; Counter:2; WaitForAck: 1;
448111: [P150] RF GW: 4 - 00,07,60,00,22,00,10,44,12,00,60,00,22,00,00,6E,FF,00,
448111: [P150] RF GW: 5 - Start waiting for ack...
448223: [P150] RF GW: 0 - sendSubscribeMessage();
448223: [P150] RF GW: 1 - SENT message (outboxQ 1): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
448223: [P150] RF GW: 2 - Bytes:18; Counter:3; WaitForAck: 1;
448224: [P150] RF GW: 3 - 00,07,60,00,22,00,10,44,13,00,60,00,22,00,00,6E,FF,00,
448224: [P150] RF GW: 4 - Start waiting for ack...
448731: [P150] RF GW: 5 - Start waiting for ack...
448851: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
448851: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
448851: [P150] RF GW: 2 - SENT message (outboxQ 1): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
448851: [P150] RF GW: 3 - Bytes:18; Counter:3; WaitForAck: 1;
448851: [P150] RF GW: 4 - 00,07,60,00,22,00,10,44,13,00,60,00,22,00,00,6E,FF,00,
448851: [P150] RF GW: 5 - Start waiting for ack...
449044: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
449044: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
449171: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
449171: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
449171: [P150] RF GW: 2 - SENT message (outboxQ 1): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
449171: [P150] RF GW: 3 - Bytes:18; Counter:3; WaitForAck: 1;
449171: [P150] RF GW: 4 - 00,07,60,00,22,00,10,44,13,00,60,00,22,00,00,6E,FF,00,
449171: [P150] RF GW: 5 - Start waiting for ack...
449484: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
449484: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
451772: WD : Uptime 8 ConnectFailures 0 FreeMem 18904 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
481769: WD : Uptime 8 ConnectFailures 0 FreeMem 19176 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
506624: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
506624: [P150] RF GW: 4 - 00,07,60,00,22,00,10,44,14,00,60,00,22,00,0C,6E,00,09,
506624: [P150] RF GW: 5 - Start waiting for ack...
506944: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
506944: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
506944: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
506944: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
506944: [P150] RF GW: 4 - 00,07,60,00,22,00,10,44,14,00,60,00,22,00,0C,6E,00,09,
506944: [P150] RF GW: 5 - Start waiting for ack...
507264: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
507264: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
507264: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
507264: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
507264: [P150] RF GW: 4 - 00,07,60,00,22,00,10,44,14,00,60,00,22,00,0C,6E,00,09,
507264: [P150] RF GW: 5 - Start waiting for ack...
507577: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
507577: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
507887: [P150] RF GW: UPDATE by PLUGIN_READ
511777: WD : Uptime 9 ConnectFailures 0 FreeMem 18608 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
541769: WD : Uptime 9 ConnectFailures 0 FreeMem 19128 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
554353: HTTP: DUCOTESTMESSAGE,7,2,1,2,1,600022000E6E000B
554358: Command: DUCOTESTMESSAGE
554361: [P150] RF GW: 0
554361: 1
554361: 2
554361: 3
554361: 4
554361: 5
554361: 6
554362: 7
554369: [P150] RF GW: Sent DUCOTESTMESSAGE to Duco Network. Databytes: 8
554370: [P150] RF GW: 0 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:60002200;Type: 7;
554370: [P150] RF GW: 1 - Bytes:18; Counter:5; WaitForAck: 0;
554370: [P150] RF GW: 2 - 00,07,60,00,22,00,10,44,15,00,60,00,22,00,0E,6E,00,0B,
554370: [P150] RF GW: 3 - Send raw Duco packet done!
567887: [P150] RF GW: UPDATE by PLUGIN_READ
```

arnemauer, 2022-08-28:
Als je de RF plugin even helemaal verwijderd. En vervolgens de volgende commando intypt:
"test_cc1101_versiontest"
"test_cc1101_partnumber"
"test_cc1101_interrupt" 
Welke informatie zie je dan terug in de logs?

BSW90, 2022-08-28:
> 

Tot op heden vertoont mijn ESP exact hetzelfde gedrag als die van TS.

```
19245118: WD : Uptime 321 ConnectFailures 0 FreeMem 19368 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19275119: WD : Uptime 321 ConnectFailures 0 FreeMem 19408 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19298961: HTTP: test_cc1101_versiontest
19305119: WD : Uptime 322 ConnectFailures 0 FreeMem 18432 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19335118: WD : Uptime 322 ConnectFailures 0 FreeMem 19176 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19365118: WD : Uptime 323 ConnectFailures 0 FreeMem 18352 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19374315: HTTP: test_cc1101_partnumber
19395119: WD : Uptime 323 ConnectFailures 0 FreeMem 19008 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
19403303: HTTP: test_cc1101_interrupt
19425119: WD : Uptime 324 ConnectFailures 0 FreeMem 19224 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

Er komt geen response op de bovenstaande commandos.
Momenteel draai ik op `ESP_Easy_mega_20220704_custom_ESP8266_4M1M`

arnemauer, 2022-08-28:
Ik ben vergeten te vermelden dat je de gateway aan een pc aan moet sluiten. En een seriele terminal moet openen om het commando uit te voeren :)

BSW90, 2022-08-28:
Geen probleem, bij deze ;)

```
>test_cc1101_versiontest
Testing CC1101 version: response 0x04:PASS

>test_cc1101_partnumber
Testing CC1101 partnumber: response 0x00:PASS

>test_cc1101_interrupt
Starting test CC1101 interruptpin
Testing CC1101 interruptpin: clock output 129 KHz, counts 31172, duration 240 ms, :PASS
```

richardvenneman, 2022-09-06:
> Als je de RF plugin even helemaal verwijderd. En vervolgens de volgende commando intypt:

Sorry ik kan dit volgen. Hoe verwijder ik de plugin helemaal? Met het "Reset" command?

> "test_cc1101_versiontest"
> "test_cc1101_partnumber"
> "test_cc1101_interrupt"
> Welke informatie zie je dan terug in de logs?

Ik begrijp dat ik deze commands op de computer moet uitvoeren, maar ik heb deze executables niet. Waar kan ik deze vinden?

arnemauer, 2022-09-20:
Woont een van jullie in het noorden? :) 

Hebben jullie ook een usb kabel die je kunt aansluiten op de ducobox? Ik zou dan graag met teamviewer met je mee willen kijken om erachter te komen waarom ze niet met elkaar communiceren.

richardvenneman, 2022-09-20:
> Woont een van jullie in het noorden? :)

Nee helaas niet (meer)

> Hebben jullie ook een usb kabel die je kunt aansluiten op de ducobox? Ik zou dan graag met teamviewer met > je mee willen kijken om erachter te komen waarom ze niet met elkaar communiceren.

Ik sta er zeker voor open om op dit issue te "pairen". Twee dingen:

1. Ik denk dat mijn Ducobox geen USB aansluiting heeft; heb je hiervoor de communication print of serial gateway nodig?
2. Ik beschik niet over een Windows computer om de Duco software te draaien, maar zou evt virtueel kunnen

arnemauer, 2022-09-20:
@richardvenneman wil je mij je adres sturen? Arne . Mauer at Gmail

BSW90, 2022-09-29:
@richardvenneman / @arnemauer zijn jullie toevallig al iets wijzer geworden?

arnemauer, 2022-09-29:
Ik heb nog geen bericht gehad. Ik ben op zoek naar iemand met dit probleem en die een usb-kabel heeft die aangesloten kan worden op de ducobox. Ik zou dan graag met teamviewer oid mee willen kijken om te zien of ik het probleem kan achterhalen.

BSW90, 2022-10-06:
Ik heb afgelopen week een USB-kabel en een kabeltje om de Serial Gateway te gebruiken geassembleerd. Beide werken zoals het hoort, ik kan verbinden via de Duco Network Tool en alle data uitlezen via de Seriele Connectie. Het enige wat nog niet werkt is de RF Gateway; de aansturing via de CC1101.  Ik sta open om een keer een teamviewer sessie in te plannen, al ben ik momenteel erg druk, misschien goed om dat even buiten deze thread om te bespreken?

arnemauer, 2022-10-17:
Met behulp van een SDR-RTL ben ik erachter gekomen dat de frequenties wat uit elkaar liggen. Dit ligt waarschijnlijk aan de CC1101 modules. Morgen ga ik dit met Bram testen op een Enri CC1101 shield.

**Ducobox**
Ingestelde frequentie: 868,29**4312** mhz
Gemeten onderste frequentie: 868,270 mhz
Gemeten bovenste frequentie: 868,310 mhz
[image]

**CO2 bedieningsschakelaar**
Ingestelde frequentie: 868,29**2328** mhz
Gemeten onderste frequentie: 868,265 mhz
Gemeten bovenste frequentie: 868,305 mhz
[image]

**Batterij bedieningsschakelaar**
Ingestelde frequentie: 868,29**6692** mhz
Gemeten onderste frequentie: 868,260 mhz
Gemeten bovenste frequentie: 868,300 mhz
[image]

**Gateway / CC1101 module**
Ingestelde frequentie: 868,29**4312** mhz
Gemeten onderste frequentie: 868,260 mhz
Gemeten bovenste frequentie: 868,300 mhz
[image]

Bram81, 2022-10-18:
Ik volgde dit item al met interesse, sinds een week of twee werkt ook mijn gateway niet meer en heb ik dezelfde issues. Veel dank voor jouw en jullie werk en ik kijk uit naar de aangepaste firmware!

Bram81, 2022-10-24:
@arnemauer, brutale vraag maar is het mogelijk om ook een bin met deze fix beschikbaar te stellen? Ik ben al twee dagen bezig om zelf een versie te compileren met arduino ide, maar ik val van de ene in de andere compilatiefout...

arnemauer, 2022-10-24:
Zeker, ik was er al mee bezig :) + een aantal extra functies. Hij komt vanavond online.

Bram81, 2022-10-24:
Top! Kijk er naar uit. Veel dank!

BSW90, 2022-10-25:
> Bij deze: [4e4d379](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/4e4d3791da5bb5204a622e2e6e2d3380ea2f31c2)

Zojuist deze versie geprobeerd, en jawel; de gateway werkt!
Thanks voor de fix @arnemauer en alle overige betrokkenen.

closed 2022-10-25 by arnemauer

Bram81, 2022-10-26:
Je bent een held! Alles werkt weer hier. Veel dank!

Bij deze: 4e4d379<https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/4e4d3791da5bb5204a622e2e6e2d3380ea2f31c2>

lasharor, 2023-04-24:
Ik krijg het helaas niet werkend met deze Setup en de laatste firmware. Ongeacht wat ik doe (afstand 5CM, 60CM en enkele meters + hoogste zendsterkte en normaal)

>test_cc1101_versiontest
Testing CC1101 version: response 0x04:PASS
>test_cc1101_partnumber
Testing CC1101 partnumber: response 0x00:PASS
>test_cc1101_interrupt
Starting test CC1101 interruptpin
Testing CC1101 interruptpin: clock output 134 KHz, counts 63951, duration 475 ms, :PASS

Iemand nog een suggestie voor iets dat ik kan proberen?