arnemauer/Ducobox-ESPEasy-Plugin #114: Gateway kan niet joinen

svh1985, 2022-08-20:
Hi,

Ik heb problemen met het Joinen van de gateway aan een DucoBox Energy Comfort WTW.
Hiervoor gebruik ik deze CC1101 en hij hangt zo ongeveer 1,5 meter van de Duco af.
[image]

Dit is de log die ik heb gekopieerd tijdens het JOIN proces. Ik heb 3x proberen te joinen.

```
301626: WD   : Uptime 5 ConnectFailures 0 FreeMem 17840 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
306405: [P150] RF GW: UPDATE by PLUGIN_READ
>> Load failed <<
331626: WD   : Uptime 6 ConnectFailures 0 FreeMem 18120 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
361626: WD   : Uptime 6 ConnectFailures 0 FreeMem 19096 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
366405: [P150] RF GW: UPDATE by PLUGIN_READ
421626: WD   : Uptime 7 ConnectFailures 0 FreeMem 19096 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
481626: WD   : Uptime 8 ConnectFailures 0 FreeMem 18744 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
486405: [P150] RF GW: UPDATE by PLUGIN_READ
528574: HTTP: JOIN
528578: Command: JOIN
528579: [P150] RF GW: Sent command for 'join' to DUCO unit
528586: [P150] RF GW: 0 - SendJoinPacket()
528586: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
528586: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
528586: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
528586: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
531020: HTTP: JOIN
531024: Command: JOIN
531025: [P150] RF GW: Sent command for 'join' to DUCO unit
531032: [P150] RF GW: 0 - SendJoinPacket()
531032: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
531032: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
531032: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
531032: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
541626: WD   : Uptime 9 ConnectFailures 0 FreeMem 19184 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
601626: WD   : Uptime 10 ConnectFailures 0 FreeMem 19104 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
606405: [P150] RF GW: UPDATE by PLUGIN_READ
631626: WD   : Uptime 11 ConnectFailures 0 FreeMem 18640 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
661626: WD   : Uptime 11 ConnectFailures 0 FreeMem 18616 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
721626: WD   : Uptime 12 ConnectFailures 0 FreeMem 18304 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
726405: [P150] RF GW: UPDATE by PLUGIN_READ
781626: WD   : Uptime 13 ConnectFailures 0 FreeMem 19128 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
786405: [P150] RF GW: UPDATE by PLUGIN_READ
901626: WD   : Uptime 15 ConnectFailures 0 FreeMem 19184 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
906405: [P150] RF GW: UPDATE by PLUGIN_READ
956644: [P150] RF GW: PLUGIN_WEBFORM_LOAD
956776: [P150] RF GW: PLUGIN_WEBFORM_LOAD
961626: WD   : Uptime 16 ConnectFailures 0 FreeMem 19184 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
962076: [P150] RF GW: EXIT PLUGIN_150
962242: [P150] RF GW: Values set from config. DeviceID: 0, networkId: 0000, radio power: Radio power 10,3 dBm (0xC1)
962245: [P150] RF GW: CC1101 868Mhz transmitter initialized
962278: [P150] RF GW: PLUGIN_WEBFORM_LOAD
962365: [P150] RF GW: UPDATE by PLUGIN_READ
974733: HTTP: JOIN
974736: Command: JOIN
974737: [P150] RF GW: Sent command for 'join' to DUCO unit
974745: [P150] RF GW: 0 - SendJoinPacket()
974746: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
974746: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
974746: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
974746: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
991626: WD   : Uptime 17 ConnectFailures 0 FreeMem 19120 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1021626: WD   : Uptime 17 ConnectFailures 0 FreeMem 19176 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1022346: [P150] RF GW: UPDATE by PLUGIN_READ
1081626: WD   : Uptime 18 ConnectFailures 0 FreeMem 19056 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1082346: [P150] RF GW: UPDATE by PLUGIN_READ
1111626: WD   : Uptime 19 ConnectFailures 0 FreeMem 19248 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1141626: WD   : Uptime 19 ConnectFailures 0 FreeMem 19120 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1142346: [P150] RF GW: UPDATE by PLUGIN_READ
1171626: WD   : Uptime 20 ConnectFailures 0 FreeMem 19248 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1201626: WD   : Uptime 20 ConnectFailures 0 FreeMem 19224 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1202346: [P150] RF GW: UPDATE by PLUGIN_READ
1231626: WD   : Uptime 21 ConnectFailures 0 FreeMem 19248 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1261626: WD   : Uptime 21 ConnectFailures 0 FreeMem 19248 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
1262346: [P150] RF GW: UPDATE by PLUGIN_READ

```

richardvenneman, 2022-08-23:
Issue lijkt op wat ik ervaar in #113

arnemauer, 2022-08-23:
De initialisatie van de cc1101 lijkt goed te gaan. Zie je wel berichten in de weblog voorbij komen als je wat knoppen indrukt op een duco afstandsbediening?

De antennes die tegenwoordig bij de cc1101 modules van aliexpress worden geleverd zijn tegenwoordig behoorlijk ruk waardoor ze heel weinig ontvangen. Wellicht dat je een andere antenne kan proberen?

arnemauer, 2022-10-25:
Waarschijnlijk lost de nieuwste firmware je probleem op: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/4e4d3791da5bb5204a622e2e6e2d3380ea2f31c2

closed 2022-10-25 by arnemauer