arnemauer/Ducobox-ESPEasy-Plugin #115: Gateway reageert niet meer met Duco box

phoenix-blue, 2022-09-12:
Hallo,

Ik heb de orginele gateway (weet niet welke versie nummer) welke km heb geüpdatet naar ESP_Easy_mega_20220704_custom_ESP8266_4M1M Jul 4 2022

Alleen reageert de Duco box niet meer op de commando's welke ik verstuur.
Heb middels de setting beide versie nummers geprobeerd evenals de diy esp8266 maar geen van 3 werken.
Gpio heb ik orgineel gehouden zoals hij al stond (gpio4 en 5)
Wat zou er aan de hand kunnen zijn?

arnemauer, 2022-09-20:
Dus totdat je de update hebt gedaan werkte de communicatie met de ducobox nog wel goed?

Zou je de blank.bin kunnen flashen en vervolgens de oude firmware? De instellingen kun je handmatig terugzetten. Werkt het dan wel weer?

Zo ja, dan weer een blank.bin flashen en daarna de nieuwe firmware. Opnieuw de instellingen handmatig terugzetten. Niet met de settings backup functie! Werkt het dan wel?

arnemauer, 2022-10-25:
Waarschijnlijk lost de nieuwste firmware je probleem op: https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/commit/4e4d3791da5bb5204a622e2e6e2d3380ea2f31c2

closed 2022-10-25 by arnemauer

phoenix-blue, 2022-10-25:
Update uitgevoerd, lijkt configuratie mee te nemen en werkt naar behoren.
Thx