arnemauer/Ducobox-ESPEasy-Plugin #120: Ventilationmode from plugin Duco ventilation remote stays on automatic as feedback

Jozziej, 2022-11-06:
Ventilationmode from plugin Duco ventilation remote stays on automatic as feedback, this worked before as I display the ventilation mode in Home Assistant

[image]
[image]

Firmware 24 October is used, rollback to 4 July firmware doesn't fix it either.

Loose plugin Duco_Ventilationmode does work, sadly that it requires to have another plugin actively sending to the controller.

arnemauer, 2022-11-06:
Make us happy with some logs :)

Jozziej, 2022-11-07:
87581606: [P154] DUCO Ext. CO2 Sensor: start read, eventid:1
87581606: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
87581607: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
87581948: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
87581959: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
87583263: [P154] DUCO Ext. CO2 Sensor: Serial reading timeout
87583264: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
87585189: [P150] RF GW: UPDATE by PLUGIN_READ
87585486: [P151] DUCO SER GW: Start readNetworkList
87585694: [P151] DUCO SER GW: Received correct response on network
87585707: [P151] DUCO SER GW: Send command successfull!
87585729: [P151] DUCO SER GW: Ducobox row found!
87585730: [P151] DUCO SER GW: New Ventilationmode value: 11
87585731: [P151] DUCO SER GW: New Countdown value: 0
87585731: [P151] DUCO SER GW: New Ventilation_Percentage value: 20
87585731: [P151] DUCO SER GW: New Current_Fan_Value value: 7
87585809: [P151] DUCO SER GW: End of message received.
87585810: [P151] DUCO SER GW: Stop reading from serial
87588819: WD   : Uptime 1460 ConnectFailures 0 FreeMem 18416 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
87591483: [P151] DUCO SER GW: Start readNetworkList
87591644: [P151] DUCO SER GW: Received correct response on network
87591659: [P151] DUCO SER GW: Send command successfull!
87591684: [P151] DUCO SER GW: Ducobox row found!
87591685: [P151] DUCO SER GW: New Ventilationmode value: 11
87591685: [P151] DUCO SER GW: New Countdown value: 0
87591686: [P151] DUCO SER GW: New Ventilation_Percentage value: 20
87591686: [P151] DUCO SER GW: New Current_Fan_Value value: 7
87591764: [P151] DUCO SER GW: End of message received.
87591764: [P151] DUCO SER GW: Stop reading from serial
87593608: [P154] DUCO Ext. CO2 Sensor: start read, eventid:1
87593608: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
87593608: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
87593950: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
87593963: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
87594709: [P154] DUCO Ext. CO2 Sensor: CO2 PPM: 414 = 414 PPM
87594716: [P154] DUCO Ext. CO2 Sensor: End of message received.
87594716: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
87597189: [P150] RF GW: UPDATE by PLUGIN_READ
87597485: [P151] DUCO SER GW: Start readNetworkList
87597648: [P151] DUCO SER GW: Received correct response on network
87597662: [P151] DUCO SER GW: Send command successfull!
87597736: [P151] DUCO SER GW: Ducobox row found!
87597736: [P151] DUCO SER GW: Status value not found.
87597737: [P151] DUCO SER GW: New Countdown value: 0
87597737: [P151] DUCO SER GW: Numeric value not found.
87597737: [P151] DUCO SER GW: Numeric value not found.
87597767: [P151] DUCO SER GW: End of message received.
87597768: [P151] DUCO SER GW: Stop reading from serial
87603483: [P151] DUCO SER GW: Start readNetworkList
87603629: [P151] DUCO SER GW: Received correct response on network
87603646: [P151] DUCO SER GW: Send command successfull!
87603669: [P151] DUCO SER GW: Ducobox row found!
87603670: [P151] DUCO SER GW: New Ventilationmode value: 11
87603670: [P151] DUCO SER GW: New Countdown value: 0
87603671: [P151] DUCO SER GW: New Ventilation_Percentage value: 20
87603671: [P151] DUCO SER GW: New Current_Fan_Value value: 7
87603754: [P151] DUCO SER GW: End of message received.
87603755: [P151] DUCO SER GW: Stop reading from serial
87605606: [P154] DUCO Ext. CO2 Sensor: start read, eventid:1
87605606: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
87605607: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
87606002: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
87606016: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
87606043: [P154] DUCO Ext. CO2 Sensor: CO2 PPM: 414 = 414 PPM
87606049: [P154] DUCO Ext. CO2 Sensor: End of message received.
87606049: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
87607144: HTTP: VENTMODE,LOW
87607146: Command: VENTMODE
87607147: [P150] RF GW: Sent command 'VENTMODE low' to DUCO unit
87607155: [P150] RF GW: 0 - requestVentilationMode();
[... 155 lines cut ...]
87642026: [P154] DUCO Ext. CO2 Sensor: CO2 PPM: 414 = 414 PPM
87642032: [P154] DUCO Ext. CO2 Sensor: End of message received.
87642033: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
87645189: [P150] RF GW: UPDATE by PLUGIN_READ
87645484: [P151] DUCO SER GW: Start readNetworkList
87645640: [P151] DUCO SER GW: Received correct response on network
87645655: [P151] DUCO SER GW: Send command successfull!
87645680: [P151] DUCO SER GW: Ducobox row found!
87645681: [P151] DUCO SER GW: New Ventilationmode value: 2
87645681: [P151] DUCO SER GW: New Countdown value: 1785
87645681: [P151] DUCO SER GW: New Ventilation_Percentage value: 40
87645682: [P151] DUCO SER GW: New Current_Fan_Value value: 26
87645760: [P151] DUCO SER GW: End of message received.
87645760: [P151] DUCO SER GW: Stop reading from serial
87648819: WD   : Uptime 1461 ConnectFailures 0 FreeMem 18528 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
87651483: [P151] DUCO SER GW: Start readNetworkList
87651635: [P151] DUCO SER GW: Received correct response on network
87651647: [P151] DUCO SER GW: Send command successfull!
87651675: [P151] DUCO SER GW: Ducobox row found!
87651676: [P151] DUCO SER GW: New Ventilationmode value: 2
87651676: [P151] DUCO SER GW: New Countdown value: 1780
87651676: [P151] DUCO SER GW: New Ventilation_Percentage value: 40
87651677: [P151] DUCO SER GW: New Current_Fan_Value value: 25
87651755: [P151] DUCO SER GW: End of message received.
87651755: [P151] DUCO SER GW: Stop reading from serial

arnemauer, 2022-11-07:
The RF-plugin doesn't receive an answer from the Ducobox. If you change the ventilation mode by pressing the blue buttons in Espeasy is the ducobox responding/changing ventilation speed?

Jozziej, 2022-11-17:
Here is a new set of logs, went from High to Medium to Low

311462845: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311462845: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
311462846: [P150] RF GW: 2 - Start waiting for ack...
311463159: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311463159: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
311472802: [P150] RF GW: UPDATE by PLUGIN_READ
311472840: [P154] DUCO Ext. CO2 Sensor: start read, eventid:1
311472840: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
311472840: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
311472987: [P151] DUCO SER GW: Serial port in use, set flag to read data later.
311473171: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
311473184: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
311474734: [P154] DUCO Ext. CO2 Sensor: Serial reading timeout
311474734: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
311474735: [P151] DUCO SER GW: Start readNetworkList
311474908: [P151] DUCO SER GW: Send command successfull!
311475322: [P151] DUCO SER GW: Received invalid response on network
311475322: [P151] DUCO SER GW: Stop reading from serial
311475323: [P151] DUCO SER GW: End of message received.
311475323: [P151] DUCO SER GW: Stop reading from serial
311478986: [P151] DUCO SER GW: Start readNetworkList
311479134: [P151] DUCO SER GW: Received correct response on network
311479138: [P151] DUCO SER GW: End of message received.
311479138: [P151] DUCO SER GW: Stop reading from serial
311480728: HTTP: VENTMODE,MIDDLE
311480729: Command: VENTMODE
311480731: [P150] RF GW: Sent command 'VENTMODE middle' to DUCO unit
311480739: [P150] RF GW: 0 - requestVentilationMode();
311480739: [P150] RF GW: 1 - sendVentilationModeMessage();
311480739: [P150] RF GW: 2 - Start waiting for ack...
311481053: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311481053: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
311481054: [P150] RF GW: 2 - Start waiting for ack...
311481372: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311481372: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
311481373: [P150] RF GW: 2 - Start waiting for ack...
311481692: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311481692: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
311481693: [P150] RF GW: 2 - Start waiting for ack...
311482046: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311482046: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
311484802: [P150] RF GW: UPDATE by PLUGIN_READ
311484840: [P154] DUCO Ext. CO2 Sensor: start read, eventid:1
311484840: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
311484840: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
311484988: [P151] DUCO SER GW: Serial port in use, set flag to read data later.
311485247: [P154] DUCO Ext. CO2 Sensor: Send command successfull!
311485294: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
311486268: [P154] DUCO Ext. CO2 Sensor: CO2 PPM: 555 = 555 PPM
311486278: [P154] DUCO Ext. CO2 Sensor: End of message received.
311486278: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
311486734: [P151] DUCO SER GW: Start readNetworkList
311486882: [P151] DUCO SER GW: Received correct response on network
311486887: [P151] DUCO SER GW: End of message received.
311486887: [P151] DUCO SER GW: Stop reading from serial
311491141: [P150] RF GW: 2 - Start waiting for ack...
311491151: [P151] DUCO SER GW: Received correct response on network
311491154: [P151] DUCO SER GW: Send command successfull!
311491191: [P151] DUCO SER GW: Ducobox row found!
311491192: [P151] DUCO SER GW: New Ventilationmode value: 2
311491192: [P151] DUCO SER GW: New Countdown value: 1795
311491193: [P151] DUCO SER GW: New Ventilation_Percentage value: 40
311491193: [P151] DUCO SER GW: New Current_Fan_Value value: 25
311491276: [P151] DUCO SER GW: End of message received.
311491276: [P151] DUCO SER GW: Stop reading from serial
311491460: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311491461: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
311491461: [P150] RF GW: 2 - Start waiting for ack...
311491774: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
311491774: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
311492056: WD   : Uptime 5192 ConnectFailures 0 FreeMem 18328 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
311496802: [P150] RF GW: UPDATE by PLUGIN_READ
311496840: [P154] DUCO Ext. CO2 Sensor: start read, eventid:1
311496840: [P154] DUCO Ext. CO2 Sensor: Read external CO2 sensor.
311496840: [P154] DUCO Ext. CO2 Sensor: Start read external sensor. NodeAddress: 3. Type: CO2
311496987: [P151] DUCO SER GW: Serial port in use, set flag to read data later.
311497177: [P154] DUCO Ext. CO2 Sensor: Command confirmed by ducobox
311497177: [P154] DUCO Ext. CO2 Sensor: End of message received.
311497177: [P154] DUCO Ext. CO2 Sensor: Stop reading from serial
311497734: [P151] DUCO SER GW: Start readNetworkList
311497886: [P151] DUCO SER GW: Received correct response on network
311497900: [P151] DUCO SER GW: Send command successfull!
311497926: [P151] DUCO SER GW: Ducobox row found!
311497927: [P151] DUCO SER GW: New Ventilationmode value: 1
311497927: [P151] DUCO SER GW: New Countdown value: 1800
311497928: [P151] DUCO SER GW: New Ventilation_Percentage value: 20
311497928: [P151] DUCO SER GW: New Current_Fan_Value value: 11

SkyThes, 2023-01-12:
Received my Ventilation Gateway today by mail and I have exact the same issue. Clicking the blue buttons in espeasy does change the modes but Ventilationmode of the RF gateway stays on Automatic. 

Log:
546509: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
546509: [P150] RF GW: 1 - Bytes:14; Counter:15; RSSI:-59 (0x1F); LQI: 0x30
546509: [P150] RF GW: 2 - 30,00,40,00,05,
546509: [P150] RF GW: 3 - Received messagetype: Normal message
546509: [P150] RF GW: 4 - Received messagetype: Normal message
548031: [P150] RF GW: 2 - 20,00,12,86,
548331: HTTP: VENTMODE,LOW
548334: Command: VENTMODE
548335: [P150] RF GW: Sent command 'VENTMODE low' to DUCO unit
548342: [P150] RF GW: 0 - requestVentilationMode();
548342: [P150] RF GW: 1 - sendVentilationModeMessage();
548342: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0010e899;Type: 7;
548342: [P150] RF GW: 3 - Bytes:18; Counter:7; WaitForAck: 1;
548342: [P150] RF GW: 4 - 00,07,00,10,E8,99,10,44,17,00,60,00,22,00,0C,6E,00,09,
548343: [P150] RF GW: 5 - Start waiting for ack...
548374: [P150] RF GW: 548348 - 548374 = 26
548374: [P150] RF GW: rxBYTES: 38 (hex) 56 (dec); getMarcState 0D
548376: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0010e899;Type: 6;
548376: [P150] RF GW: 1 - Bytes:9; Counter:1; RSSI:-96 (0xD4); LQI: 0x31
548376: [P150] RF GW: 2 - 
548660: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
548660: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
548661: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0010e899;Type: 7;
548661: [P150] RF GW: 3 - Bytes:18; Counter:7; WaitForAck: 1;
548661: [P150] RF GW: 4 - 00,07,00,10,E8,99,10,44,17,00,60,00,22,00,0C,6E,00,09,
548661: [P150] RF GW: 5 - Start waiting for ack...
548673: [P150] RF GW: 548666 - 548673 = 7
548674: [P150] RF GW: rxBYTES: 41 (hex) 65 (dec); getMarcState 11
548675: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
548675: [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-59 (0x1F); LQI: 0x2E
548675: [P150] RF GW: 2 - 20,00,52,01,
548675: [P150] RF GW: 3 - Received messagetype: Normal message
548675: [P150] RF GW: 4 - Received messagetype: Normal message
548725: [P150] RF GW: RX FIFO OVERFLOW -> idle, flush RX FIFO and switch to RX state.
548988: [P150] RF GW: 6 - Received ACK for message outboxQ 0 we've sent.
549325: [P150] RF GW: 549318 - 549325 = 7
549326: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
549327: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
549327: [P150] RF GW: 1 - Bytes:14; Counter:3; RSSI:-59 (0x1F); LQI: 0x2E
549327: [P150] RF GW: 2 - 30,00,40,00,05,
549327: [P150] RF GW: 3 - Received messagetype: Normal message
549327: [P150] RF GW: 4 - Received messagetype: Normal message
549645: [P150] RF GW: 549639 - 549645 = 6
549646: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
549647: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
549647: [P150] RF GW: 1 - Bytes:14; Counter:3; RSSI:-58 (0x20); LQI: 0x2F
549647: [P150] RF GW: 2 - 30,00,40,00,05,
549648: [P150] RF GW: 3 - Received messagetype: Normal message
549648: [P150] RF GW: 4 - Received messagetype: Normal message
549977: [P150] RF GW: 549962 - 549976 = 14
549977: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
549979: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
549979: [P150] RF GW: 1 - Bytes:14; Counter:3; RSSI:-59 (0x1F); LQI: 0x2D
549979: [P150] RF GW: 2 - 30,00,40,00,05,
549979: [P150] RF GW: 3 - Received messagetype: Normal message
549979: [P150] RF GW: 4 - Received messagetype: Normal message
550715: [P150] RF GW: UPDATE by PLUGIN_READ
550930: [P150] RF GW: 550928 - 550930 = 2
550931: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
550932: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
550932: [P150] RF GW: 1 - Bytes:14; Counter:4; RSSI:-58 (0x20); LQI: 0x2E
550932: [P150] RF GW: 2 - 30,00,40,00,05,
550932: [P150] RF GW: 3 - Received messagetype: Normal message
550933: [P150] RF GW: 4 - Received messagetype: Normal message
551270: [P150] RF GW: 551250 - 551270 = 20
551271: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
551272: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
[... 45 lines cut ...]
554797: [P150] RF GW: 2 - 20,00,12,84,
554797: [P150] RF GW: 3 - Received messagetype: Normal message
554797: [P150] RF GW: 4 - Received messagetype: Normal message
555129: [P150] RF GW: 555115 - 555128 = 13
555129: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
555131: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
555131: [P150] RF GW: 1 - Bytes:13; Counter:7; RSSI:-59 (0x1F); LQI: 0x2F
555131: [P150] RF GW: 2 - 20,00,12,84,
555131: [P150] RF GW: 3 - Received messagetype: Normal message
555131: [P150] RF GW: 4 - Received messagetype: Normal message
555766: [P150] RF GW: 4 - Received messagetype: Normal message
556083: [P150] RF GW: 556081 - 556083 = 2
[... cut ...]
556085: [P150] RF GW: 4 - Received messagetype: Normal message
556423: [P150] RF GW: 556403 - 556423 = 20
556424: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
556425: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
556425: [P150] RF GW: 1 - Bytes:14; Counter:8; RSSI:-58 (0x20); LQI: 0x2E
556425: [P150] RF GW: 2 - 30,00,40,00,05,
556425: [P150] RF GW: 3 - Received messagetype: Normal message
556425: [P150] RF GW: 4 - Received messagetype: Normal message

arnemauer, 2023-01-12:
What is the distance between the gateway and your ducobox? The gateway isn't receiving any rf message from the ducobox....

paddenstoeltje, 2025-12-23:
Hi I'm having the same problem, however I'm receiving data:

1343356: [P150] RF GW: 1343344 - 1343356 = 12
1343358: [P150] RF GW: rxBYTES: 30 (hex) 48 (dec); getMarcState 0D
1344022: [P150] RF GW: 1344010 - 1344022 = 12
1344023: [P150] RF GW: rxBYTES: 31 (hex) 49 (dec); getMarcState 0D
1344566: [P150] RF GW: 1344552 - 1344566 = 14
1344567: [P150] RF GW: rxBYTES: 32 (hex) 50 (dec); getMarcState 0D

TX works well, RX worked for about 1.5 year but now suuddenly it does not work anymore?

Any thoughts?