arnemauer/Ducobox-ESPEasy-Plugin #121: When ESP8266 reboots, the Duco goes to High Fan RPM

Jozziej, 2022-11-07:
Is there a way to make sure that when the ESP8266 reboots, the Duco does not go to High?
Simply I always have the Duco at Permanent Low, which is changed to 20% in the Duco box itself.

arnemauer, 2022-11-07:
After reboot it is assumed that the fan is running in automatic mode. 
https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/blob/master/_P150_DucoRFGateway.ino#L21

The gateway will send a subscribe message to the Ducobox to get the current ventilation mode.

Jozziej, 2022-11-17:
For some reason the ventilation does go to High every time the esp8266 lost connection or did a reboot. Is there a way to force a certain ventilation value, like permanent low?

arnemauer, 2022-11-17:
Do you use mqtt? Could you try to disable that controller and reboot? Is de ducobox still changing ventilation mode to high after a reboot?

Jozziej, 2022-11-17:
Yes I use MQTT to Home Assistant, when I disable the MQTT controller and reboot the ESP, the Duco box does not change the Fan to High. As soon as I enable the Mqtt controller again, instand change to Fan High.

arnemauer, 2022-11-17:
It is not the plugin changing the ventilationmode. If you open the mqtt controller configuration page, do you have the retain checkbox on?

Jozziej, 2022-11-17:
Yes

[image]

arnemauer, 2022-11-17:
Uncheck it...

Jozziej, 2022-11-17:
Tried that, and makes no difference

Jozziej, 2023-04-16:
Hi Arne you can close this issue.
For some reason the Mqtt broker retained the Ventilation High cmd. Solved that.

closed 2023-04-16 by Jozziej