arnemauer/Ducobox-ESPEasy-Plugin #123: Gateway kan geen connectie maken

pmkruyen, 2022-11-19:
Het lukt mij helaas niet om Ducobox te ‘joinen’. 

Na op de ‘joint’ knop gedrukt te hebben, zie ik wel:
- messagereceivedcounter: 25
- messagesentcounter: 5

Maar de network id en device address staan nog steeds op 0.

Ik maak gebruik van de laatste firmware versie. 

Zouden jullie mij kunnen vertellen hoe ik verder kan debuggen? How ik kan zien dat er iets mis is gegaan in de hardware bijvoorbeeld? Als meer informatie nodig is dan hoor ik het graag. 

Dank. 

Ps ik heb de stappen gevolgd in de wiki (zie mijn andere issue). 

```
[1051857](tel:1051857): WD : Uptime 18 ConnectFailures 0 FreeMem 17096 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
[1074371](tel:1074371): [P150] RF GW: [1074354 - 1074371](tel:1074354 - 1074371) = 17
[1074372](tel:1074372): [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
[1074374](tel:1074374): [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d88f;Type: 7;
[1074374](tel:1074374): [P150] RF GW: 1 - Bytes:13; Counter:15; RSSI:-86 (0xE7); LQI: 0x51
[1074374](tel:1074374): [P150] RF GW: 2 - 20,00,52,00,
[1074374](tel:1074374): [P150] RF GW: 3 - Ignore message, not our networkID.
[1080602](tel:1080602): [P150] RF GW: UPDATE by PLUGIN_READ
[1081857](tel:1081857): WD : Uptime 18 ConnectFailures 0 FreeMem 17120 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
[1094700](tel:1094700): HTTP: JOIN
[1094701](tel:1094701): Command: JOIN
[1094702](tel:1094702): [P150] RF GW: Sent command for 'join' to DUCO unit
[1094709](tel:1094709): [P150] RF GW: 0 - SendJoinPacket()
[1094709](tel:1094709): [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
[1094709](tel:1094709): [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
[1094709](tel:1094709): [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
[1094709](tel:1094709): [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
[1108001](tel:1108001): [P150] RF GW: [1107984 - 1108001](tel:1107984 - 1108001) = 17
[1108002](tel:1108002): [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
[1108004](tel:1108004): [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d6e;Type: 7;
[1108004](tel:1108004): [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-93 (0xDA); LQI: 0x34
[1108004](tel:1108004): [P150] RF GW: 2 - 20,00,52,00,
[1108004](tel:1108004): [P150] RF GW: 3 - Ignore message, not our networkID.
```

Quitastic, 2022-11-24:
Ik heb de print (Enri CC1101 shield) en precies hetzelfde probleem en ook hetzelfde gedaan als @pmkruyen. Ik heb verschillende firmwares geprobeerd. Ook firmwares van voor de wijzigingen. Ok heb ik de instellingen voor de 8.4 versie van de gateway aan en uit gehad. 

Ik heb de Installer mode geactiveerd op de duco box zelf of op een afstand bediening. Ik heb ook eerst de stroom eraf gehaald, 10 seconden gewacht en de stroom er weer op gezet. Nadat de Duco klaar was met blazen heb ik de installer mode geactiveerd.

Ik heb een DucoBox Silent met serie nummer 1818356-180502-049.

Ik heb geen Frequentie meter tot mijn beschikking (SDR-RTL).

De laatste controles uit dit issua #113 heb ik nog niet gedaan. Ik ben wel in het bezit van een seriele kabel die goed werkt.

arnemauer, 2022-11-24:
Welke firmware versie hebben jullie gebruikt?
Wat is de afstand tussen de esp8266 en de ducobox?

Quitastic, 2022-11-25:
Ik heb 20221024 Firmware c785203 24-10-2022 (ESPEasy mega-20220427).bin erop gezet. Afstand tot mijn Ducobox is ongeveer 1.5 meter zonder obstakels. Heb ook met voorgaande versies getest van voor de frequentie wijzegingen.

Ik heb hier nog een andere Wemos D1 mini en een NodeMCU liggen waar ik de tests nogmaals mee ga doen. Degene waarop ik het allemaal op heb getest is van AliExpress en ik kan niet goed achterhalen of er een goede volt regulator op zit. Ik houd jullie op de hoogte.

pmkruyen, 2022-11-25:
👋

Ik heb eerst gewerkt met de kaatste firmware. Ik zit toevallig net te testen met de allereerste firmware. Helaas ook zonder resultaat. 

Mijn Wemos D1 komt van een electro zaak, mijn shield van Amazon (maar weet niet wat dit zegt over de kwaliteit). Ik probeer het te koppelen aan de Ducobox Silent Connect.

Mijn afstand tot de box is ook 1,5 m. 

Hieronder de log met de eerste firmware, probeer snel opnieuw te testen met de laatste firmware. 

====

`361937: WD   : Uptime 6 ConnectFailures 0 FreeMem 12312 WiFiStatus 3
371617: [P150] DUCO RF GW: TIMER => 371616 371614 2
371617: [P150] DUCO RF GW: Start of RF signal received
371618: [P150] DUCO RF GW: 371618 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d88f;
371619: [P150] DUCO RF GW: 371618 - Type:0; Bytes:10; Counter:9; RSSI:-85 (0xEA); 
371619: [P150] DUCO RF GW: 371618 - FF,00,
371619: [P150] DUCO RF GW: 371618 - Ignore message, not our networkID.
374152: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
374152: [P150] DUCO RF GW:  State: 0.00
376436: [P150] DUCO RF GW: TIMER => 376436 376419 17
376437: [P150] DUCO RF GW: Start of RF signal received
376438: [P150] DUCO RF GW: 376438 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d88f;
376439: [P150] DUCO RF GW: 376438 - Type:7; Bytes:13; Counter:10; RSSI:-85 (0xE9); 
376439: [P150] DUCO RF GW: 376438 - FF,20,00,52,00,
376439: [P150] DUCO RF GW: 376438 - Ignore message, not our networkID.
384316: [P150] DUCO RF GW: TIMER => 384316 384298 18
384317: [P150] DUCO RF GW: Start of RF signal received
384318: [P150] DUCO RF GW: 384318 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d839;
384319: [P150] DUCO RF GW: 384318 - Type:0; Bytes:10; Counter:7; RSSI:-94 (0xD8); 
384319: [P150] DUCO RF GW: 384318 - FF,00,
384319: [P150] DUCO RF GW: 384318 - Ignore message, not our networkID.
389116: [P150] DUCO RF GW: TIMER => 389116 389103 13
389117: [P150] DUCO RF GW: Start of RF signal received
389118: [P150] DUCO RF GW: 389118 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d839;
389119: [P150] DUCO RF GW: 389118 - Type:7; Bytes:13; Counter:8; RSSI:-94 (0xD8); 
389119: [P150] DUCO RF GW: 389118 - FF,20,00,52,00,
389119: [P150] DUCO RF GW: 389118 - Ignore message, not our networkID.
391937: WD   : Uptime 7 ConnectFailures 0 FreeMem 11944 WiFiStatus 3
394377: [P150] DUCO RF GW: TIMER => 394376 394369 7
394377: [P150] DUCO RF GW: Start of RF signal received
394379: [P150] DUCO RF GW: 394378 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d759;
394379: [P150] DUCO RF GW: 394378 - Type:7; Bytes:13; Counter:14; RSSI:-84 (0xEB); 
394379: [P150] DUCO RF GW: 394378 - FF,20,00,52,00,
394380: [P150] DUCO RF GW: 394378 - Ignore message, not our networkID.
409338: [P150] DUCO RF GW: TIMER => 409338 409331 7
409338: [P150] DUCO RF GW: Start of RF signal received
409340: [P150] DUCO RF GW: 409339 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d6e;
409340: [P150] DUCO RF GW: 409339 - Type:7; Bytes:13; Counter:12; RSSI:-92 (0xDB); 
409341: [P150] DUCO RF GW: 409340 - FF,20,00,52,00,
409341: [P150] DUCO RF GW: 409340 - Ignore message, not our networkID.
421937: WD   : Uptime 7 ConnectFailures 0 FreeMem 11784 WiFiStatus 3
434152: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
434152: [P150] DUCO RF GW:  State: 0.00
442237: [P150] DUCO RF GW: TIMER => 442236 442235 1
442237: [P150] DUCO RF GW: Start of RF signal received
442237: [P150] DUCO RF GW: Ignoring RF noise
451937: WD   : Uptime 8 ConnectFailures 0 FreeMem 11792 WiFiStatus 3
458997: [P150] DUCO RF GW: TIMER => 458996 458986 10
458997: [P150] DUCO RF GW: Start of RF signal received
458997: [P150] DUCO RF GW: Ignoring RF noise
462156: [P150] DUCO RF GW: TIMER => 462156 462145 11
462157: [P150] DUCO RF GW: Start of RF signal received
462157: [P150] DUCO RF GW: Ignoring RF noise
481937: WD   : Uptime 8 ConnectFailures 0 FreeMem 11744 WiFiStatus 3
482938:  Webserver args: 0: 'cmd' length: 4
482961: Command: JOIN
482962: [P150] DUCO RF GW: Sent command for 'join' to DUCO unit
482968: [P150] DUCO RF GW: 482963 - SendJoinPacket()
[... 316 lines cut ...]
846493: [P150] DUCO RF GW: 846492 - Type:0; Bytes:10; Counter:9; RSSI:-65 (0x13); 
846493: [P150] DUCO RF GW: 846492 - FF,00,
846493: [P150] DUCO RF GW: 846492 - Ignore message, not our networkID.
850070: [P150] DUCO RF GW: TIMER => 850070 850062 8
850071: [P150] DUCO RF GW: Start of RF signal received
850071: [P150] DUCO RF GW: Ignoring RF noise
851311: [P150] DUCO RF GW: TIMER => 851310 851294 16
851311: [P150] DUCO RF GW: Start of RF signal received
851313: [P150] DUCO RF GW: 851312 - Received message: SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003b6e8;
851313: [P150] DUCO RF GW: 851312 - Type:7; Bytes:13; Counter:10; RSSI:-66 (0x11); 
851313: [P150] DUCO RF GW: 851312 - FF,20,00,52,00,
851313: [P150] DUCO RF GW: 851312 - Ignore message, not our networkID.
854152: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
854152: [P150] DUCO RF GW:  State: 0.00
866231: [P150] DUCO RF GW: TIMER => 866230 866220 10
866231: [P150] DUCO RF GW: Start of RF signal received
866231: [P150] DUCO RF GW: Ignoring RF noise
871891: [P150] DUCO RF GW: TIMER => 871890 871890 0
871891: [P150] DUCO RF GW: Start of RF signal received
871891: [P150] DUCO RF GW: Ignoring RF noise
871936: WD   : Uptime 15 ConnectFailures 0 FreeMem 11464 WiFiStatus 3
901936: WD   : Uptime 15 ConnectFailures 0 FreeMem 11464 WiFiStatus 3
914152: [P150] DUCO RF GW: UPDATE by PLUGIN_READ
914152: [P150] DUCO RF GW:  State: 0.00
931936: WD   : Uptime 16 ConnectFailures 0 FreeMem 11440 WiFiStatus 3`

pmkruyen, 2022-11-28:
👋 Dit is wat ik zie met de laatste firmware (2 x joint gedrukt). 

```
139446: [P150] RF GW: 139434 - 139446 = 12
139447: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
139449: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003b6dc;Type: 0;
139449: [P150] RF GW: 1 - Bytes:10; Counter:1; RSSI:-95 (0xD6); LQI: 0x39
139449: [P150] RF GW: 2 - 00,
139449: [P150] RF GW: 3 - Ignore message, not our networkID.
142728: [P150] RF GW: 142715 - 142727 = 12
142728: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
142730: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d6e;Type: 7;
142730: [P150] RF GW: 1 - Bytes:13; Counter:13; RSSI:-81 (0xF2); LQI: 0x36
142730: [P150] RF GW: 2 - 20,00,52,00,
142730: [P150] RF GW: 3 - Ignore message, not our networkID.
144254: [P150] RF GW: 144239 - 144253 = 14
144254: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
144256: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003b6dc;Type: 7;
144256: [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-95 (0xD5); LQI: 0x34
144256: [P150] RF GW: 2 - 20,00,52,00,
144256: [P150] RF GW: 3 - Ignore message, not our networkID.
151634: WD   : Uptime 3 ConnectFailures 0 FreeMem 15648 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
180379: [P150] RF GW: UPDATE by PLUGIN_READ
181634: WD   : Uptime 3 ConnectFailures 0 FreeMem 15064 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
186211: HTTP: JOIN
186212: Command: JOIN
186213: [P150] RF GW: Sent command for 'join' to DUCO unit
186220: [P150] RF GW: 0 - SendJoinPacket()
186220: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
186220: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
186220: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
186220: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
207952: [P150] RF GW: 207944 - 207952 = 8
207953: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
207954: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d61;Type: 0;
207955: [P150] RF GW: 1 - Bytes:10; Counter:6; RSSI:-78 (0xF7); LQI: 0x34
207955: [P150] RF GW: 2 - 00,
207955: [P150] RF GW: 3 - Ignore message, not our networkID.
211634: WD   : Uptime 4 ConnectFailures 0 FreeMem 15560 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
212886: [P150] RF GW: 212885 - 212886 = 1
212887: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
212889: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d61;Type: 7;
212889: [P150] RF GW: 1 - Bytes:13; Counter:7; RSSI:-80 (0xF3); LQI: 0x33
212889: [P150] RF GW: 2 - 20,00,52,00,
212889: [P150] RF GW: 3 - Ignore message, not our networkID.
215772: [P150] RF GW: 215767 - 215771 = 4
215772: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
215774: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003d758;Type: 7;
215774: [P150] RF GW: 1 - Bytes:13; Counter:15; RSSI:-95 (0xD6); LQI: 0x2E
215774: [P150] RF GW: 2 - 20,00,52,00,
215774: [P150] RF GW: 3 - Ignore message, not our networkID.
226489: [P150] RF GW: 226489 - 226489 = 0
226490: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
226491: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d45;Type: 0;
226492: [P150] RF GW: 1 - Bytes:10; Counter:13; RSSI:-93 (0xD9); LQI: 0x33
226492: [P150] RF GW: 2 - 00,
226492: [P150] RF GW: 3 - Ignore message, not our networkID.
231299: [P150] RF GW: 231294 - 231299 = 5
231300: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
231302: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00036d45;Type: 7;
231302: [P150] RF GW: 1 - Bytes:13; Counter:14; RSSI:-94 (0xD8); LQI: 0x33
231302: [P150] RF GW: 2 - 20,00,52,00,
231302: [P150] RF GW: 3 - Ignore message, not our networkID.
240385: [P150] RF GW: UPDATE by PLUGIN_READ
241634: WD   : Uptime 4 ConnectFailures 0 FreeMem 15728 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
247576: [P150] RF GW: PLUGIN_WEBFORM_LOAD
271634: WD   : Uptime 5 ConnectFailures 0 FreeMem 16176 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
272197: HTTP: JOIN
272199: Command: JOIN
272200: [P150] RF GW: Sent command for 'join' to DUCO unit
[... 3 lines cut ...]
272208: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
272208: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
277462: [P150] RF GW: 277461 - 277462 = 1
277463: [P150] RF GW: rxBYTES: 0F (hex) 15 (dec); getMarcState 0D
277465: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003b6e8;Type: 7;
277465: [P150] RF GW: 1 - Bytes:12; Counter:5; RSSI:-80 (0xF3); LQI: 0x35
277465: [P150] RF GW: 2 - 10,00,51,
277465: [P150] RF GW: 3 - Ignore message, not our networkID.
282271: [P150] RF GW: 282266 - 282271 = 5
282272: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
282274: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003b6e8;Type: 0;
282274: [P150] RF GW: 1 - Bytes:10; Counter:6; RSSI:-77 (0xF9); LQI: 0x31
282274: [P150] RF GW: 2 - 00,
282274: [P150] RF GW: 3 - Ignore message, not our networkID.
[... cut ...]
287080: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0003b6e8;Type: 7;
287080: [P150] RF GW: 1 - Bytes:13; Counter:7; RSSI:-83 (0xED); LQI: 0x31
287080: [P150] RF GW: 2 - 20,00,52,00,
287080: [P150] RF GW: 3 - Ignore message, not our networkID.
300379: [P150] RF GW: UPDATE by PLUGIN_READ
301634: WD   : Uptime 5 ConnectFailures 0 FreeMem 15904 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

@arnemauer kun je ons op een of andere wijze verder helpen?

arnemauer, 2022-11-28:
@pmkruyen  wat is jou ducobox network id?

pmkruyen, 2022-11-28:
Hoe kan ik dat achterhalen?

Jover68, 2022-11-30:
Het probleem van pmkruyen had ik een aantal maanden terug na een update ook. In de LOG zag je dan "not our networkID" staan. In de DUCO ventilation remote instellingen zou het ID uit zichzelf worden gevuld maar deze bleef na de update leeg bij mij. Gelukkig had ik screenshots gemaakt, handmatig de oude ingevoerd en werkt tot nu toe perfect. Ik zie in de log van pmkruyen meestal Ntwrk:0003b6dc voorbij komen, Je kan het allicht proberen.

pmkruyen, 2022-11-30:
Dank @Jover68, ga het proberen! 

Wat moet ik dan bij “device address” invullen?

arnemauer, 2022-11-30:
Handmatig invullen van het networkid en device id kan alleen wanneer de gateway al eerder succesvol is aangemeld. Ik kijk vanmiddag naar de logs.

Quitastic, 2022-11-30:
Ik heb wat verder onderzoek gedaan, goede resultaten, maar het werkt nog steeds niet. Zie hieronder:

130655155: Command: CC1101STATUS
130655157: [P150] RF GW: getMarcState: 0D

130691083: Command: RXBYTES
130691085: [P150] RF GW: rxBYTES: 00 (hex) 0 (dec)

>test_cc1101_versiontest
Testing CC1101 version: response 0x04:PASS

>test_cc1101_partnumber
Testing CC1101 partnumber: response 0x00:PASS

Starting test CC1101 interruptpin
Testing CC1101 interruptpin: clock output 132 KHz, counts 55050, duration 414 ms, :PASS

Kan het iets te maken hebben met dat ik een vrij nieuwe Duco heb met DUCO_BMB_BASIC-2v3 markering op het board zelf? Ik heb echt letterlijk hetzelfde als wat hieronder staat, behalve dan de network id die bij mij anders is.

> 👋 Dit is wat ik zie met de laatste firmware (2 x joint gedrukt).
>
> ```
> 139446: [P150] RF GW: 139434 - 139446 = 12
> 139447: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
> 139449: [P150] RF [...]

Ik heb ook de voltage regulator op de wemos d1 mini gecontroleerd en ben er zeker van dat deze niet het probleem is. Dit is bij mij een de=a1d.

Het enige wat ik nog niet heb gedaan is de gehele Duco factory resetten en alles opnieuw aanmelden. Als dit mis gaat is mijn vrouw weer niet blij :P Ik heb constant de instructies gevolgd zoals in dit filmpje https://www.youtube.com/watch?v=HheLI0m5qOM. Echter druk ik dan niet op een afstandbediening maar op de join knop in ESPEasy.

Quitastic, 2022-11-30:
Hoe kan je het netwerk id van je Duco Box achterhalen?
Hoe kan je het Device Address en Network Id van een bestaande device achterhalen? Ik vermoed dat Ntwrk:0003b6e8 het network id is. Kan ik als device address het volgnummer in de Duco Network Tool invullen?

Heb je nog meer informatie van ons nodig?

arnemauer, 2022-11-30:
@Quitastic  Je kan het id achterhalen door een aantal keer op een knop van de afstandsbediening te drukken terwijl je in de weblog kijkt.

pmkruyen, 2022-11-30:
😁 zou cool zijn 🥳

Kan ik nu wel of niet het network id handmatig instellen? 

En waar kan ik het device id vinden?

Ik ga het morgen proberen. 

Groet

PS. Ik realiseerde mij ook dat de draadloze afstandsbediening steeds op de box ligt. Ik weet niet of die de verbinding verstoort? Ik haal er morgen de batterij uit. M

arnemauer, 2022-11-30:
> Kan ik nu wel of niet het network id handmatig instellen?
Nee

> En waar kan ik het device id vinden?
In de duco network tool maar daar moet je een kabel voor hebben. Maar nogmaals, dat heeft geen zin als de gateway zich niet al eerder succesvol heeft aangemeld.

> PS. Ik realiseerde mij ook dat de draadloze afstandsbediening steeds op de box ligt. Ik weet niet of die de verbinding verstoort? 
Maakt niet uit :)

pmkruyen, 2022-11-30:
Ok. Interessant dat het signaal zo slecht is.

Ik probeer het morgen nogmaals. Ook nog eens met oudere firmware. Ga niet opgeven. 😉

Hoor het ook graag als jullie nog tips hebben.

Quitastic, 2022-11-30:
Haha nee we wonen niet naast elkaar. Ik had een stukje van zijn log gequote.

Ok ik heb het network id van  mn RF afstandbediening in de badkamer achterhaald. Met de network tool kan ik zien dat het device 3 is. Zie hieronder:

```
75438: [P150] RF GW: 75430 - 75437 = 7
75438: [P150] RF GW: rxBYTES: 0C (hex) 12 (dec); getMarcState 0D
75440: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:00052cd6;Type: 6;
75440: [P150] RF GW: 1 - Bytes:9; Counter:1; RSSI:-58 (0x20); LQI: 0x7F
75440: [P150] RF GW: 2 -
75440: [P150] RF GW: 3 - Ignore message, not our networkID.
87912: [P150] RF GW: 87909 - 87912 = 3
87913: [P150] RF GW: rxBYTES: 0C (hex) 12 (dec); getMarcState 0D
87915: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:00052cd6;Type: 6;
87915: [P150] RF GW: 1 - Bytes:9; Counter:6; RSSI:-60 (0x1D); LQI: 0x7F
87915: [P150] RF GW: 2 -
87915: [P150] RF GW: 3 - Ignore message, not our networkID.
```

 Heb deze handmatig overgenomen in ESPEasy en op submit gedrukt. De Duco box reageert hier ook niet op als ik verschillende standen probeer aan te zetten. Zie hiervoor de log hieronder:

```
90421: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
90421: [P150] RF GW: 4 - 00,07,00,05,2C,D6,18,46,14,00,60,00,22,00,0E,6E,00,0B,
90421: [P150] RF GW: 5 - Start waiting for ack...
90729: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
90729: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
90729: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:00052cd6;Type: 7;
90729: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
90729: [P150] RF GW: 4 - 00,07,00,05,2C,D6,18,46,14,00,60,00,22,00,0E,6E,00,0B,
90729: [P150] RF GW: 5 - Start waiting for ack...
91049: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
91049: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
91049: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:00052cd6;Type: 7;
91049: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
91049: [P150] RF GW: 4 - 00,07,00,05,2C,D6,18,46,14,00,60,00,22,00,0E,6E,00,0B,
91050: [P150] RF GW: 5 - Start waiting for ack...
91384: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
91384: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
91384: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:00052cd6;Type: 7;
91385: [P150] RF GW: 3 - Bytes:18; Counter:4; WaitForAck: 1;
91385: [P150] RF GW: 4 - 00,07,00,05,2C,D6,18,46,14,00,60,00,22,00,0E,6E,00,0B,
91385: [P150] RF GW: 5 - Start waiting for ack...
91696: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
91696: [P150] RF GW: 1 - CheckforAck: no ack received, cancel retrying.
```

pmkruyen, 2022-11-30:
We zitten wel gezellig met dezelfde uitdaging @Quitastic, en een vrouw die zich afvraagt waarom de ventilatie steeds aan en uit gaat 😉.

pmkruyen, 2022-12-03:
Helaas nig steeds geen succes. Ik ga deze week met firmware van vorig jaar testen. Bij jou wel gelujt @Quitastic ?

Quitastic, 2022-12-04:
Ik heb deze van Henri => https://www.marktplaats.nl/v/hobby-en-vrije-tijd/elektronica-componenten/m1910607707-itho-of-duco-rf-wifi-gateway-inclusief-behuizing?correlationId=1c47ef58-e98a-41d8-95e7-9269b97a259b 

Verder heb ik een Duco Silent met het board type DUCO_BMB_BASIC-2v3 en serie nummer 1818356-180502-049.

pmkruyen, 2022-12-05:
Helaas ook geen succes met oudere versies van de firmware. 

Zie hier… mocht het nodig zijn dat ik nog andere foto’s toevoeg, hoor ik het graag. Dank voor het meedenken. 

[image]
[image]
[image]
[image]

arnemauer, 2022-12-05:
Ik denk dat de antenne/module combinatie zorgt voor een afwijking in de frequentie. Ik heb een testfirmware waarmee we de frequentie stap voor stap kunnen aanpassen. Heeft een van jullie ergens op een avond een uurtje tijd waarbij ik via teamviewer het een en ander kan testen?

arnemauer, 2022-12-06:
> Jazeker! Leuk. Hoe kan ik je een PM sturen?

Wil je mij een e-mail sturen op cxws7kt4qhig@wegwerpmailadres.nl (binnen 24 uur)?

pmkruyen, 2022-12-06:
👍

pmkruyen, 2022-12-06:
Sorry, helaas mailadres bestaat niet. Kun je een nieuwe aanmaken?

Quitastic, 2022-12-06:
Heren wat fijn dat jullie er zoveel tijd insteken. Dankjewel!. Ik heb precies dezelfde hardware als pmkruyen dus ik hoor wel wat er uit jullie meeting komt. Als jullie het resultaat hier willen posten zou dat geniaal zijn.

pmkruyen, 2022-12-06:
@arnemauer, je kunt mij hier bereiken: garnaal_dromedaris09@icloud.com

Quitastic, 2022-12-22:
Hallo. Zijn jullie al wat wijzer geworden? Hebben jullie al overleg gehad en wat dingen kunnen controleren?

dgokoel, 2023-01-07:
Hoi, ik heb zojuist de Gateway RF binnengekregen, exact dezelfde issue met het verbinden als hierboven. Er zijn wat kleine wijzigingen in de schermpjes, zoals het kunnen kiezen van je device. Ik heb eerst alles gedaan volgens de stappen. Daarna in de plugin gekozen voor de 8.4 gateway. Helaas zonder succes. Toen ik het na allerlei opties het nog steeds niet voor elkaar kreeg, heb ik de laatste firmware geupload, was succesvol, opnieuw aangesloten, maar daarna helaas helemaal geen verbinding meer met de gateway. Ik zie ook geen nieuwe toewijzing van een ander IP adres. Iemand een idee wat ik zou kunnen doen?

arnemauer, 2023-01-07:
@dgokoel de antenne in de gateway is afgestemd  op de configuratie van de cc1101. Wat was de afstand tussen je ducobox en de gateway? Gebruik je een kwalitatief goede kabel en adapter?

dgokoel, 2023-01-07:
Dank voor je reactie, ik gebruik een Samsung oplader (van 2A) met kwaliteitskabel en was de box aan instellen in een kamer ernaast, ik schat 3 meter. Ik ga het morgen proberen van een langere afstand.

dgokoel, 2023-01-09:
Hoi Arne, helaas hetzelfde resultaat. Ook met een andere lader die nog sterker is en op een grotere afstand. Ik heb de log ook aangezet terwijl ik de stekker van de Duco eruit en erin heb gedaan. Het is nogal wat tekst dus heb hem als bijlage erbij gedaan. Kun je daar iets mee? Dank in ieder geval!

[log_090123.txt](https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/files/10372532/log_090123.txt)

arnemauer, 2023-01-09:
> Hoi Arne, helaas hetzelfde resultaat. Ook met een andere lader die nog sterker is en op een grotere afstand. Ik heb de log ook aangezet terwijl ik de stekker van de Duco eruit en erin heb gedaan. [...]

De gateway lijkt de berichten van zowel je ducobox als een bedieningsschakelaar prima te ontvangen. Wat is de afstand tussen de gateway en je ducobox? 2 meter tussen de gateway en je ducobox is prima. 

Zou je de weblog open kunnen houden en vervolgens de join knop in willen drukken? Ducobox natuurlijk in installer mode zetten :)

dgokoel, 2023-01-09:
Na join, krijg ik geen network id of device address (en wederkerend connect failures in de log). Je kunt waarschijnlijk zien dat ik de duco weer opnieuw heb gestart en ik toch nog maar een keer op join heb gedrukt.

```
91605: WD   : Uptime 2 ConnectFailures 0 FreeMem 17368 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
101075: [P150] RF GW: PLUGIN_WEBFORM_LOAD
113490: [P150] RF GW: PLUGIN_WEBFORM_LOAD
120353: [P150] RF GW: UPDATE by PLUGIN_READ
121606: WD   : Uptime 2 ConnectFailures 0 FreeMem 17520 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
151605: WD   : Uptime 3 ConnectFailures 0 FreeMem 17536 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
180353: [P150] RF GW: UPDATE by PLUGIN_READ
181606: WD   : Uptime 3 ConnectFailures 0 FreeMem 17768 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
181745: [P150] RF GW: 181729 - 181745 = 16
181746: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
181747: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 0;
181747: [P150] RF GW: 1 - Bytes:10; Counter:1; RSSI:-79 (0xF5); LQI: 0x30
181747: [P150] RF GW: 2 - 00,
181747: [P150] RF GW: 3 - Ignore message, not our networkID.
181885: [P150] RF GW: 181884 - 181885 = 1
181886: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
181887: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 0;
181887: [P150] RF GW: 1 - Bytes:10; Counter:1; RSSI:-72 (0x04); LQI: 0x32
181887: [P150] RF GW: 2 - 00,
181887: [P150] RF GW: 3 - Ignore message, not our networkID.
186531: [P150] RF GW: 186531 - 186531 = 0
186532: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
186533: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 7;
186533: [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-80 (0xF4); LQI: 0x32
186533: [P150] RF GW: 2 - 20,00,52,01,
186533: [P150] RF GW: 3 - Ignore message, not our networkID.
191344: [P150] RF GW: 191333 - 191344 = 11
191345: [P150] RF GW: rxBYTES: 1C (hex) 28 (dec); getMarcState 0D
191346: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
191346: [P150] RF GW: 1 - Bytes:13; Counter:1; RSSI:-78 (0xF7); LQI: 0x31
191346: [P150] RF GW: 2 - 20,00,52,01,
191346: [P150] RF GW: 3 - Ignore message, not our networkID.
192333: [P150] RF GW: 2 - 
192333: [P150] RF GW: 3 - Ignore message, not our networkID.
192434: [P150] RF GW: RX FIFO OVERFLOW -> idle, flush RX FIFO and switch to RX state.
192651: [P150] RF GW: 192648 - 192651 = 3
192652: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
192653: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
192653: [P150] RF GW: 1 - Bytes:14; Counter:2; RSSI:-77 (0xF9); LQI: 0x2E
192653: [P150] RF GW: 2 - 30,00,40,00,05,
192653: [P150] RF GW: 3 - Ignore message, not our networkID.
192671: [P150] RF GW: 192654 - 192671 = 17
192672: [P150] RF GW: rxBYTES: 1F (hex) 31 (dec); getMarcState 0D
192673: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 6;
192673: [P150] RF GW: 1 - Bytes:9; Counter:2; RSSI:-73 (0x02); LQI: 0x2F
192673: [P150] RF GW: 2 - 
192673: [P150] RF GW: 3 - Ignore message, not our networkID.
192991: [P150] RF GW: 192971 - 192991 = 20
192992: [P150] RF GW: rxBYTES: 26 (hex) 38 (dec); getMarcState 0D
192993: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 7;
192993: [P150] RF GW: 1 - Bytes:16; Counter:12; RSSI:-70 (0x09); LQI: 0x33
192993: [P150] RF GW: 2 - 50,00,41,00,05,00,11,
192993: [P150] RF GW: 3 - Ignore message, not our networkID.
193276: [P150] RF GW: 193273 - 193276 = 3
193277: [P150] RF GW: rxBYTES: 26 (hex) 38 (dec); getMarcState 0D
193278: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 7;
193278: [P150] RF GW: 1 - Bytes:16; Counter:12; RSSI:-71 (0x06); LQI: 0x30
193279: [P150] RF GW: 2 - 50,00,41,00,05,00,11,
193279: [P150] RF GW: 3 - Ignore message, not our networkID.
194903: [P150] RF GW: 2 - 
194903: [P150] RF GW: 3 - Ignore message, not our networkID.
195221: [P150] RF GW: 195202 - 195221 = 19
195222: [P150] RF GW: rxBYTES: 33 (hex) 51 (dec); getMarcState 0D
195223: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 7;
195223: [P150] RF GW: 1 - Bytes:16; Counter:13; RSSI:-76 (0xFC); LQI: 0x30
195223: [P150] RF GW: 2 - 50,00,41,00,05,00,11,
195223: [P150] RF GW: 3 - Ignore message, not our networkID.
[... 166 lines cut ...]
420353: [P150] RF GW: UPDATE by PLUGIN_READ
421605: WD   : Uptime 7 ConnectFailures 0 FreeMem 19144 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
424065: [P150] RF GW: PLUGIN_WEBFORM_LOAD
451605: WD   : Uptime 8 ConnectFailures 0 FreeMem 19144 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
460524: [P150] RF GW: Sent command for 'join' to DUCO unit
460531: [P150] RF GW: 0 - SendJoinPacket()
460531: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
460531: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
460531: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
460531: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
[... cut ...]
460779: [P150] RF GW: 4 - processJoin2Packet()
460779: [P150] RF GW: 5 - joinPacketValidNetworkId()
460779: [P150] RF GW: 6 - SendJoin3Packet: INVALID join2 packet received!
480353: [P150] RF GW: UPDATE by PLUGIN_READ
481605: WD   : Uptime 8 ConnectFailures 0 FreeMem 19168 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
491530: [P150] RF GW: PLUGIN_WEBFORM_LOAD
493589: [P150] RF GW: PLUGIN_WEBFORM_LOAD
511623: WD   : Uptime 9 ConnectFailures 0 FreeMem 19144 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

arnemauer, 2023-01-09:
Wat is de afstand tussen de gateway en de ducobox? Je bedieningschakelaar heeft namelijk een beter bereik dan je ducobox. Daarom gaat je bedieningschakelaar de aanmelding van de gateway afhandelen in plaats van de Ducobox. Daar is de code op dit moment nog niet op aangepast. Je zal hem daarom dichter bij je ducobox moeten zetten.

dgokoel, 2023-01-09:
De afstand is nu de kabellengte van de serial verbinding + de  meegeleverde rj11 kabel; aangezien die straks ook verbonden zal worden. Dit krijg ik nu na een join:

```
562684: [P150] RF GW: RX FIFO OVERFLOW -> idle, flush RX FIFO and switch to RX state.
562943: [P150] RF GW: 562938 - 562943 = 5
562944: [P150] RF GW: rxBYTES: 17 (hex) 23 (dec); getMarcState 0D
562945: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
562945: [P150] RF GW: 1 - Bytes:13; Counter:1; RSSI:-42 (0x41); LQI: 0x31
562945: [P150] RF GW: 2 - 20,00,52,00,
562945: [P150] RF GW: 3 - Ignore message, not our networkID.
563331: [P150] RF GW: 563270 - 563331 = 61
563332: [P150] RF GW: rxBYTES: 00 (hex) 0 (dec); getMarcState 0D
563332: [P150] RF GW: !!! Interrupt received, but no new message?!
563332: [P150] RF GW: rxBYTES: 00 (hex) 0 (dec); getMarcState 0D
571606: WD : Uptime 10 ConnectFailures 0 FreeMem 19056 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
600355: [P150] RF GW: UPDATE by PLUGIN_READ
601612: WD : Uptime 10 ConnectFailures 0 FreeMem 19056 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
631640: WD : Uptime 11 ConnectFailures 0 FreeMem 19056 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
660354: [P150] RF GW: UPDATE by PLUGIN_READ
661606: WD : Uptime 11 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
682394: HTTP: JOIN
682395: Command: JOIN
682396: [P150] RF GW: Sent command for 'join' to DUCO unit
682403: [P150] RF GW: 0 - SendJoinPacket()
682403: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
682403: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
682403: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
682403: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
684083: [P150] RF GW: RX FIFO OVERFLOW -> idle, flush RX FIFO and switch to RX state.
684367: [P150] RF GW: 684349 - 684367 = 18
684368: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
684369: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
684369: [P150] RF GW: 1 - Bytes:13; Counter:4; RSSI:-42 (0x41); LQI: 0x30
684369: [P150] RF GW: 2 - 20,00,12,00,
684370: [P150] RF GW: 3 - Ignore message, not our networkID.
684687: [P150] RF GW: 684669 - 684687 = 18
684688: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
684689: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
684689: [P150] RF GW: 1 - Bytes:13; Counter:4; RSSI:-42 (0x40); LQI: 0x30
684689: [P150] RF GW: 2 - 20,00,12,00,
684689: [P150] RF GW: 3 - Ignore message, not our networkID.
686853: [P150] RF GW: 686851 - 686853 = 2
686854: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
686855: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
686855: [P150] RF GW: 1 - Bytes:13; Counter:5; RSSI:-42 (0x41); LQI: 0x2E
686856: [P150] RF GW: 2 - 20,00,12,00,
686856: [P150] RF GW: 3 - Ignore message, not our networkID.
691607: WD : Uptime 12 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
713458: [P150] RF GW: 713444 - 713458 = 14
713459: [P150] RF GW: rxBYTES: 0C (hex) 12 (dec); getMarcState 0D
713460: [P150] RF GW: 0 - RECEIVED message (inboxQ 0): SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 6;
713460: [P150] RF GW: 1 - Bytes:9; Counter:15; RSSI:-42 (0x40); LQI: 0x30
713460: [P150] RF GW: 2 -
713460: [P150] RF GW: 3 - Ignore message, not our networkID.
714675: [P150] RF GW: 714665 - 714675 = 10
714676: [P150] RF GW: rxBYTES: 00 (hex) 0 (dec); getMarcState 0D
714676: [P150] RF GW: !!! Interrupt received, but no new message?!
714676: [P150] RF GW: rxBYTES: 00 (hex) 0 (dec); getMarcState 0D
720355: [P150] RF GW: UPDATE by PLUGIN_READ
721606: WD : Uptime 12 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
751606: WD : Uptime 13 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
780353: [P150] RF GW: UPDATE by PLUGIN_READ
781606: WD : Uptime 13 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
811606: WD : Uptime 14 ConnectFailures 0 FreeMem 19176 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
840353: [P150] RF GW: UPDATE by PLUGIN_READ
841606: WD : Uptime 14 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
871607: WD : Uptime 15 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
900353: [P150] RF GW: UPDATE by PLUGIN_READ
901607: WD : Uptime 15 ConnectFailures 0 FreeMem 19080 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```

dgokoel, 2023-01-09:
Bijzonder, want hij stond wel op installermode, nu nog een keer opnieuw gestart en daarna handmatig de installermode aangezet op de box (groen knipperlampje): zorgt voor deze output:

```
278448: HTTP: JOIN
278449: Command: JOIN
278450: [P150] RF GW: Sent command for 'join' to DUCO unit
278456: [P150] RF GW: 0 - SendJoinPacket()
278457: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
278457: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
278457: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
278457: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
300666: [P150] RF GW: UPDATE by PLUGIN_READ
```

dgokoel, 2023-01-09:
Als je hier niks in ziet, of hij lijkt niet op installer mode te staan ga ik nog even iets anders uitzoeken. En misschien de duco network tool is bij de hand nemen.

arnemauer, 2023-01-09:
Dat is te weinig log om er iets zinnigs over te zeggen :)

dgokoel, 2023-01-19:
Hoi @arnemauer, zou je kunnen uitleggen wat je precies hebt gedaan bij @pmkruyen? Misschien dat de gateway hier ook zal werken. Ik heb geprobeerd de gateway op 5cm van de box te plaatsen maar dan heb ik geen wifi verbinding meer, dus kan ik ook niet joinenof iets uitlezen. Het lijkt er op dat hij inderdaad minimaal op 2,5m van de box moet liggen om de wifi te verbinden. Dus ik ben benieuwd! Groet!

Quitastic, 2023-01-25:
Hey  @arnemauer,

Kan je ons uitleggen wat je hebt gedaan om het voor pmkruyen te laten werken? Wat kunnen wij doen om dit ook te laten werken? Ik heb dezelfde antenne als pmkruyen. Ik hoor graag wat de oplossing is. Ik sta ook open voor een andere antenne die beter is. Help ons alsjeblieft :)

arnemauer, 2023-01-26:
Eigenlijk heb ik helemaal niets gedaan om het op te lossen :) 
De antenne die standaard bij losse CC1101 modules wordt geleverd is ontzettend slecht. Zo slecht dat de module van @pmkruyen pas communicatie met de ducobox had op 5 cm van de ducobox en RF power op maximaal. 
Met die combinatie zou je ongemerkt zomaar een flinke stoorzender kunnen zijn voor de radiosignalen in de directe omgeving.

@dgokoel @Quitastic Welke module gebruiken jullie?

dgokoel, 2023-01-26:
hoi @arnemauer, dank voor je reactie! Ik weet eerlijk gezegd niet welke antenne ik heb, ik heb de RF gateway van je ontvangen waar hij al in zat. Zou je dit kunnen nagaan?  Ik zag wel dat er op een van de device instellingen een tabblad was waarbij ik een sterker signaal kon kiezen. Ik zal dat ook eens proberen.

arnemauer, 2023-01-27:
@dgokoel Je moet "Ventilation gateway V8.3 and lower" kiezen ;) 8.4 is een hele kleine batch geweest met andere hardware.

Quitastic, 2023-01-31:
Dankjewel voor je reactie. Ik ga deze dingen proberen. Moet nog wel even een nieuwe antenne bestellen, dus dat duurt wel even voor ik weer een reactie kan plaatsen.

matgeroe, 2023-06-01:
Hi, Ik heb denk ik een soortgelijk probleem.

Ik heb:
* een Duco silentbox, op het printje staat DUCO_BMB_BASIC-2v3
* twee RF afstandsbedieningen, gemonteerd in de muur
* nodemcu met cc1101 868MHz

Ik woon in een appartementencomplex, iedereen heeft een silent box, mogelijk niet precies dezelfde versie, maar het kan zijn dat er daar wat verkeer van bij zit.

Ik denk dat m'n netwerk id: 0008dad3 is, dat zie ik vaak voorbijkomen als ik op knopjes duw.  
In de log zijn verschillende voorbeelden waar ik op join druk in de interface. Eigenlijk krijg ik altijd `INVALID join2 packet received!`.  
Volgens mij heb ik ergens gezien wat de verschillende ids betekenen, (0 voor broadcast, 1 voor duco, de rest voor afstandsbediening).
Waar ik mijn gateway ook plaats, dichtbij de kast of verder weg, ik zie nooit verkeer van `SRC:1`, het is altijd geproxied via `SRC:2` of `SRC:3`, `ORG.SRC:1` staat er dan netjes bij. (vandaar ook het invalid join packet las ik). Als ik alles afmeld, dan is het volledig stil.  
Het lijkt alsof de gateway geen verkeer van de Duco ziet, maar alleen van de afstandsbedieningen. 
Ik heb verschillende firmwares geprobeerd, en zit nu op die van 20230518.
Wellicht mankeert er iets aan mijn CC1101, waardoor hij niet zendt? Of heeft mijn Duco net een ander bordje, waardoor diens frequentie afwijkt en de box en de gateway langs elkaar praten?

Enig inzicht wordt zeer gewaardeerd!

<details>
<summary>captures</summary> 
117420: DEBUG: String size:7168<br>
117476: static_file: esp.css<br>
117504: HTML : Request file esp.css<br>
117607: [P150] RF GW: UPDATE by PLUGIN_READ<br>
124683: [P150] RF GW: 124667 - 124683 = 16<br>
124683: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D<br>
124685: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:2; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008dad3;Type: 7;<br>
124685: [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-66 (0x10); LQI: 0x7B<br>
124685: [P150] RF GW: 2 - 20,00,52,00,<br>
124685: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
127489: DEBUG: String size:4608<br>
127541: static_file: esp.css<br>
127542: HTML : Request file esp.css<br>
139016: WD   : Uptime 2 ConnectFailures 0 FreeMem 11832 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init<br>
139904: LoopStats: shortestLoop: 64 longestLoop: 2530371 avgLoopDuration: 90.74 loopCounterMax: 468750 loopCounterLast: 320695<br>
139905: Scheduler stats: (called/tasks/max_length/idle%) 1296709/7673/7/91.01<br>
154875: [P150] RF GW: 154872 - 154875 = 3<br>
154875: [P150] RF GW: rxBYTES: 14 (hex) 20 (dec); getMarcState 0D<br>
154877: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:2; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008dad3;Type: 7;<br>
154877: [P150] RF GW: 1 - Bytes:17; Counter:12; RSSI:-64 (0x15); LQI: 0x7F<br>
154877: [P150] RF GW: 2 - 60,00,22,00,2D,77,00,0E,<br>
154877: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
154895: [P150] RF GW: 154892 - 154895 = 3<br>
154895: [P150] RF GW: rxBYTES: 0C (hex) 12 (dec); getMarcState 0D<br>
154896: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:2; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008dad3;Type: 6;<br>
154896: [P150] RF GW: 1 - Bytes:9; Counter:12; RSSI:-66 (0x10); LQI: 0x7B<br>
154896: [P150] RF GW: 2 - <br>
154897: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
156348: [P150] RF GW: 156342 - 156347 = 5<br>
156348: [P150] RF GW: rxBYTES: 19 (hex) 25 (dec); getMarcState 0D<br>
156350: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:2; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008dad3;Type: 7;<br>
156350: [P150] RF GW: 1 - Bytes:13; Counter:15; RSSI:-66 (0x10); LQI: 0x72<br>
156350: [P150] RF GW: 2 - 20,00,12,0D,<br>
156350: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
156368: [P150] RF GW: 156349 - 156367 = 18<br>
156368: [P150] RF GW: rxBYTES: 0C (hex) 12 (dec); getMarcState 0D<br>
156369: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008dad3;Type: 6;<br>
156369: [P150] RF GW: 1 - Bytes:9; Counter:15; RSSI:-64 (0x14); LQI: 0x7F<br>
156369: [P150] RF GW: 2 - <br>
156369: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
156707: [P150] RF GW: 156692 - 156706 = 14<br>
156707: [P150] RF GW: rxBYTES: 0C (hex) 12 (dec); getMarcState 0D<br>
156708: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0008dad3;Type: 6;<br>
156708: [P150] RF GW: 1 - Bytes:9; Counter:1; RSSI:-67 (0x0E); LQI: 0x7F<br>
156709: [P150] RF GW: 2 - <br>
156709: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
156727: [P150] RF GW: 156713 - 156727 = 14<br>
156727: [P150] RF GW: rxBYTES: 14 (hex) 20 (dec); getMarcState 0D<br>
156729: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0008dad3;Type: 7;<br>
[... 453 lines cut ...]
185089: Serve 304: 31283112 esp.css<br>
185089: HTML : Request file esp.css<br>
185155: [P150] RF GW: 185146 - 185155 = 9<br>
185155: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D<br>
185157: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:2; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008dad3;Type: 2;<br>
185157: [P150] RF GW: 1 - Bytes:13; Counter:15; RSSI:-62 (0x18); LQI: 0x7F<br>
185157: [P150] RF GW: 2 - 00,00,7C,3E,<br>
185157: [P150] RF GW: 3 - Received messagetype: JOIN2<br>
185157: [P150] RF GW: 4 - processJoin2Packet()<br>
185157: [P150] RF GW: 5 - joinPacketValidNetworkId()<br>
[... cut ...]
190332: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008baab;Type: 0;<br>
190332: [P150] RF GW: 1 - Bytes:10; Counter:10; RSSI:-77 (0xF9); LQI: 0x5D<br>
190332: [P150] RF GW: 2 - 00,<br>
190332: [P150] RF GW: 3 - Ignore message, not our networkID.<br>
214484: WD   : Uptime 4 ConnectFailures 0 FreeMem 12488 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init<br>
------------------------------------------------------------------------------------------------------

</details>