arnemauer/Ducobox-ESPEasy-Plugin #125: Build mislukt door fout

BasvanH, 2023-01-05:
Ik wil zelf een build maken omdat ik wat zaken aan ESPEasy wil toevoegen.

Ik heb een Wemos D1 mini kloon, dat is een ESP8266 met 4MB flash.

De stappen die ik heb ondernomen zijn als volgt:
- Laatste ESPEasy gedownload.
- Alle Ducobox plugins in de /src map bij de andere plugins geplaatst.
- De Duco map met de lib bestanden in /lib geplaatst (dus /lib/Duco/<bestanden>)
- In platformio.ini default_envs op normal_ESP8266_4M1M
- Dan Project Tasks normal_ESP8266_4M1M Build.

Tijdens de build krijg ik op een gegeven moment de volgende foutmelding:
```
C:/Arduino/Sketches/D1/ESPEasy-mega/src/_P151_DucoSerialGateway.ino: In function 'boolean Plugin_151(byte, EventStruct*, String&)':
C:/Arduino/Sketches/D1/ESPEasy-mega/src/_P151_DucoSerialGateway.ino:196:77: error: no matching function for call to 'addSelector_Head_reloadOnChange(String)'
     addSelector_Head_reloadOnChange(PCONFIG_LABEL(P151_CONFIG_HARDWARE_TYPE));
                                                                             ^
C:/Arduino/Sketches/D1/ESPEasy-mega/src/_P151_DucoSerialGateway.ino:196:77: note: candidates are:
In file included from src\_Plugin_Helper.h:55:0,
                 from C:/Arduino/Sketches/D1/ESPEasy-mega/src/ESPEasy.ino:102:
src\src/WebServer/Markup.h:99:6: note: void addSelector_Head_reloadOnChange(const __FlashStringHelper*)
 void addSelector_Head_reloadOnChange(const __FlashStringHelper * id);
      ^
src\src/WebServer/Markup.h:99:6: note:   no known conversion for argument 1 from 'String' to 'const __FlashStringHelper*'
src\src/WebServer/Markup.h:103:6: note: void addSelector_Head_reloadOnChange(const String&, const __FlashStringHelper*, bool, const String&)
 void addSelector_Head_reloadOnChange(const String& id,
      ^
src\src/WebServer/Markup.h:103:6: note:   candidate expects 4 arguments, 1 provided
src\src/WebServer/Markup.h:112:6: note: void addSelector_Head_reloadOnChange(const String&, const __FlashStringHelper*, const String&, bool, const String&)
 void addSelector_Head_reloadOnChange(const String& id,
      ^
src\src/WebServer/Markup.h:112:6: note:   candidate expects 5 arguments, 1 provided
*** [.pio\build\normal_ESP8266_4M1M\src\ESPEasy.ino.cpp.o] Error 1
```

Ik heb een Clean en Clean All geprobeerd, maar dat helpt niet. Wat kan hiervan de oorzaak zijn?

arnemauer, 2023-01-06:
Ik gebruik ESPEasy_mega-20220427 voor de huidige binaries. Je kan in de github van espeasy even kijken welke verandering er zijn gemaakt in "addSelector_Head_reloadOnChange" na 27 april 2022. Op basis daarvan kun je de plugin aanpassen. Als je dan een pull request in deze github doet dan kunnen we nieuwe binaries maken :)

BasvanH, 2023-01-18:
Deze [regel](https://github.com/letscontrolit/ESPEasy/blob/72ce17fb0dd15c892e00083296b56c6f963b4f94/src/src/WebServer/Markup.h#L100) is in deze [commit](https://github.com/letscontrolit/ESPEasy/commit/01bee9eadc53f12b0a6ab59bb7a7ded6046a31be) uitgezet, als ik die weer activeer werkt de Ducobox plugin build weer. Het lukt mij alleen niet om jouw code zo te veranderen dat het weer gaat werken met de nieuwe ESPEasy code, met die regel dus uitgeschakeld. Zou je er na deze bevindingen zelf even naar willen kijken?

arnemauer, 2023-05-18:
Ik heb een nieuwe build gemaakt op basis van ESPEasy 15-05-2023 :) 
Bedankt voor je werk!

closed 2023-05-18 by arnemauer