arnemauer/Ducobox-ESPEasy-Plugin #131: Invalid JSON when using ESP Easy REST API

samvdb, 2023-01-09:
The ventilation gateway has been running for a long time now, however something changed and i have no clue what happened.

I use the following api call to get info: http://x.x.x.x/json?tasknr=1

This now gives me:

```json
{
"TaskValues": [
{"ValueNumber":1,
"Name":"Ventilationmode",
"NrDecimals":0,
"Value":1
}],
"TTL":60000,
"DataAcquisition": [
{"Controller":1,
"IDX":0,
"Enabled":"false"
},
{"Controller":2,
"IDX":0,
"Enabled":"false"
},
{"Controller":3,
"IDX":0,
"Enabled":"false"
}],
"TaskInterval":60,
"Type":"DUCO ventilation remote",
"TaskName":"test",
"TaskDeviceNumber":150,
"TaskEnabled":"true",
"TaskNumber":4
}
"TTL":60000
}
```

Not the last TTL that is missing a colon, the json is now invalid. I tried the obvious reboot but to no avail. Any help on this one?

```
Firmware
--
Build:⋄ | 20104  - Mega
System Libraries:⋄ | ESP82xx Core 3d128e5c, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | AFM 2c20f83: Jun  4 2020
Plugin Count:⋄ | 54 [Normal]
Build Time:⋄ | Jun  4 2020 12:57:34
Binary Filename:⋄ | Self built!

```

closed 2023-01-14 by samvdb

arnemauer, 2023-01-14:
Did you fixed it? Please share with us :)

samvdb, 2023-01-14:
No, this is an espeasy issue so it seems. I ordered new gateway with new
firmware, will test again.