arnemauer/Ducobox-ESPEasy-Plugin #132: No feedback from ducobox when changing ventilationmode

arnemauer, 2023-01-12:
Received my Ventilation Gateway today by mail and I have exact the same issue. Clicking the blue buttons in espeasy does change the modes but Ventilationmode of the RF gateway stays on Automatic. 

Log:
546509: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
546509: [P150] RF GW: 1 - Bytes:14; Counter:15; RSSI:-59 (0x1F); LQI: 0x30
546509: [P150] RF GW: 2 - 30,00,40,00,05,
546509: [P150] RF GW: 3 - Received messagetype: Normal message
546509: [P150] RF GW: 4 - Received messagetype: Normal message
548031: [P150] RF GW: 2 - 20,00,12,86,
548331: HTTP: VENTMODE,LOW
548334: Command: VENTMODE
548335: [P150] RF GW: Sent command 'VENTMODE low' to DUCO unit
548342: [P150] RF GW: 0 - requestVentilationMode();
548342: [P150] RF GW: 1 - sendVentilationModeMessage();
548342: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0010e899;Type: 7;
548342: [P150] RF GW: 3 - Bytes:18; Counter:7; WaitForAck: 1;
548342: [P150] RF GW: 4 - 00,07,00,10,E8,99,10,44,17,00,60,00,22,00,0C,6E,00,09,
548343: [P150] RF GW: 5 - Start waiting for ack...
548374: [P150] RF GW: 548348 - 548374 = 26
548374: [P150] RF GW: rxBYTES: 38 (hex) 56 (dec); getMarcState 0D
548376: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0010e899;Type: 6;
548376: [P150] RF GW: 1 - Bytes:9; Counter:1; RSSI:-96 (0xD4); LQI: 0x31
548376: [P150] RF GW: 2 - 
548660: [P150] RF GW: 0 - CheckforAck: check if 300ms is verstreken
548660: [P150] RF GW: 1 - CheckforAck: still waiting for ACK. Sending message again...
548661: [P150] RF GW: 2 - SENT message (outboxQ 0): SRC:2; DEST:1; ORG.SRC:2; ORG.DEST:1; Ntwrk:0010e899;Type: 7;
548661: [P150] RF GW: 3 - Bytes:18; Counter:7; WaitForAck: 1;
548661: [P150] RF GW: 4 - 00,07,00,10,E8,99,10,44,17,00,60,00,22,00,0C,6E,00,09,
548661: [P150] RF GW: 5 - Start waiting for ack...
548673: [P150] RF GW: 548666 - 548673 = 7
548674: [P150] RF GW: rxBYTES: 41 (hex) 65 (dec); getMarcState 11
548675: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
548675: [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-59 (0x1F); LQI: 0x2E
548675: [P150] RF GW: 2 - 20,00,52,01,
548675: [P150] RF GW: 3 - Received messagetype: Normal message
548675: [P150] RF GW: 4 - Received messagetype: Normal message
548725: [P150] RF GW: RX FIFO OVERFLOW -> idle, flush RX FIFO and switch to RX state.
548988: [P150] RF GW: 6 - Received ACK for message outboxQ 0 we've sent.
549325: [P150] RF GW: 549318 - 549325 = 7
549326: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
549327: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
549327: [P150] RF GW: 1 - Bytes:14; Counter:3; RSSI:-59 (0x1F); LQI: 0x2E
549327: [P150] RF GW: 2 - 30,00,40,00,05,
549327: [P150] RF GW: 3 - Received messagetype: Normal message
549327: [P150] RF GW: 4 - Received messagetype: Normal message
549645: [P150] RF GW: 549639 - 549645 = 6
549646: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
549647: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
549647: [P150] RF GW: 1 - Bytes:14; Counter:3; RSSI:-58 (0x20); LQI: 0x2F
549647: [P150] RF GW: 2 - 30,00,40,00,05,
549648: [P150] RF GW: 3 - Received messagetype: Normal message
549648: [P150] RF GW: 4 - Received messagetype: Normal message
549977: [P150] RF GW: 549962 - 549976 = 14
549977: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
549979: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
549979: [P150] RF GW: 1 - Bytes:14; Counter:3; RSSI:-59 (0x1F); LQI: 0x2D
549979: [P150] RF GW: 2 - 30,00,40,00,05,
549979: [P150] RF GW: 3 - Received messagetype: Normal message
549979: [P150] RF GW: 4 - Received messagetype: Normal message
550715: [P150] RF GW: UPDATE by PLUGIN_READ
550930: [P150] RF GW: 550928 - 550930 = 2
550931: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
550932: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
550932: [P150] RF GW: 1 - Bytes:14; Counter:4; RSSI:-58 (0x20); LQI: 0x2E
550932: [P150] RF GW: 2 - 30,00,40,00,05,
550932: [P150] RF GW: 3 - Received messagetype: Normal message
550933: [P150] RF GW: 4 - Received messagetype: Normal message
551270: [P150] RF GW: 551250 - 551270 = 20
551271: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
551272: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
[... 47 lines cut ...]
554797: [P150] RF GW: 4 - Received messagetype: Normal message
555129: [P150] RF GW: 555115 - 555128 = 13
555129: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
555131: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
555131: [P150] RF GW: 1 - Bytes:13; Counter:7; RSSI:-59 (0x1F); LQI: 0x2F
555131: [P150] RF GW: 2 - 20,00,12,84,
555131: [P150] RF GW: 3 - Received messagetype: Normal message
555131: [P150] RF GW: 4 - Received messagetype: Normal message
555766: [P150] RF GW: 4 - Received messagetype: Normal message
556083: [P150] RF GW: 556081 - 556083 = 2
556084: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
556085: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
556085: [P150] RF GW: 1 - Bytes:14; Counter:8; RSSI:-58 (0x20); LQI: 0x30
556085: [P150] RF GW: 2 - 30,00,40,00,05,
556085: [P150] RF GW: 3 - Received messagetype: Normal message
556085: [P150] RF GW: 4 - Received messagetype: Normal message
556423: [P150] RF GW: 556403 - 556423 = 20
556424: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
556425: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:2; ORG.SRC:1; ORG.DEST:2; Ntwrk:0010e899;Type: 7;
556425: [P150] RF GW: 1 - Bytes:14; Counter:8; RSSI:-58 (0x20); LQI: 0x2E
556425: [P150] RF GW: 2 - 30,00,40,00,05,
556425: [P150] RF GW: 3 - Received messagetype: Normal message
556425: [P150] RF GW: 4 - Received messagetype: Normal message

_Originally posted by @SkyThes in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/120#issuecomment-1380586158_

arnemauer, 2023-01-12:
You have no other RF duco remotes?

SkyThes, 2023-01-12:
@arnemauer 
Thanks a lot for your quick reply! After flashing the blank image and then the firmware image it is working as expected. So looks indeed like some sort of corruption. 👍

closed 2023-01-12 by arnemauer