arnemauer/Ducobox-ESPEasy-Plugin #134: DEBUG log ondersteuning voor laatste ESPEasy build

BasvanH, 2023-01-18:
Tijdens het oplossen van #125 liep ik tegen het probleem aan dat ik build fouten kreeg op `LOG_LEVEL_DEBUG`. Sinds de [november 5 release van ESPEasy](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20221105) is er een global variable gemaakt genaamd `BUILD_NO_DEBUG`. Deze bepaald of een debug log gedeelte wel of niet in de build moet worden meegenomen om zo ruimte te besparen. Standaard wordt met de ESP8266 1M en 4M build de debug logs blokken **niet** meer meegenomen. Dit betekend dat de log plekken in de Ducobox plugin moeten worden nagelopen of deze echt voor debug zijn of toch meer informationeel. Voor informationeel moet het dan gewijzigd worden naar `LOG_LEVEL_INFO`, en voor echt debug (wat dus standaard niet meer in de build zal meegaan) zal er een wrap rondom dat gedeelte heen moeten.

```
# ifndef BUILD_NO_DEBUG
// De debug log code
# endif // ifndef BUILD_NO_DEBUG
```

Ik kan niet beoordelen wel stuk dus INFO of DEBUG moet zijn, dus @arnemauer als je wilt dat de Ducobox plugin ondersteuning voor die versie en later gaat krijgen is dit een aandachtspunt.

Ik heb simpelweg alles naar LOG_LEVEL_INFO omgezet en dat werkt prima. De plugin werkt dan (na toepassing van #125) helemaal zonder problemen.

arnemauer, 2023-01-18:
Super! Ik ga dit weekend ernaar kijken.

BasvanH, 2024-10-28:
Zou je hier nog eens naar willen kijken en mogelijk je code willen updaten / aanpassen naar de laatste ESPEasy build?