arnemauer/Ducobox-ESPEasy-Plugin #135: Enri CC1101 shield - geen koppeling met Ducobox

arnemauer, 2023-01-27:
@Quitastic
Ik heb de print (Enri CC1101 shield) en precies hetzelfde probleem en ook hetzelfde gedaan als @pmkruyen. Ik heb verschillende firmwares geprobeerd. Ook firmwares van voor de wijzigingen. Ok heb ik de instellingen voor de 8.4 versie van de gateway aan en uit gehad. 

Ik heb de Installer mode geactiveerd op de duco box zelf of op een afstand bediening. Ik heb ook eerst de stroom eraf gehaald, 10 seconden gewacht en de stroom er weer op gezet. Nadat de Duco klaar was met blazen heb ik de installer mode geactiveerd.

Ik heb een DucoBox Silent met serie nummer 1818356-180502-049.

Ik heb geen Frequentie meter tot mijn beschikking (SDR-RTL).

De laatste controles uit dit issua #113 heb ik nog niet gedaan. Ik ben wel in het bezit van een seriele kabel die goed werkt.

_Originally posted by @Quitastic in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/123#issuecomment-1326869275_

JaccovdS, 2023-02-16:
@Quitastic toevallig al resultaten met de andere antenne?

arnemauer, 2023-02-16:
@Quitastic kan ik via teamviewer een keer met je meekijken en wat zaken testen?

Quitastic, 2023-03-12:
Hallo!. Ik hield dit topic niet in de gaten, alleen die andere. Ik heb de antennes inmiddels binnen gekregen, maar moet er nog eentje op solderen. Het ligt allemaal klaar, maar zit nu midden in een kleine verbouwing. 

Ik hoop binnenkort een gaatje te vinden en dit te testen. Ik houd jullie op de hoogte, maar verwacht niet volgende week al resultaat.