arnemauer/Ducobox-ESPEasy-Plugin #136: Ventilation Gateway - geen koppeling met Ducobox

arnemauer, 2023-01-27:
Na join, krijg ik geen network id of device address (en wederkerend connect failures in de log). Je kunt waarschijnlijk zien dat ik de duco weer opnieuw heb gestart en ik toch nog maar een keer op join heb gedrukt.

<details>
  <summary>LOG</summary>
```
91605: WD   : Uptime 2 ConnectFailures 0 FreeMem 17368 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
101075: [P150] RF GW: PLUGIN_WEBFORM_LOAD
113490: [P150] RF GW: PLUGIN_WEBFORM_LOAD
120353: [P150] RF GW: UPDATE by PLUGIN_READ
121606: WD   : Uptime 2 ConnectFailures 0 FreeMem 17520 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
151605: WD   : Uptime 3 ConnectFailures 0 FreeMem 17536 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
180353: [P150] RF GW: UPDATE by PLUGIN_READ
181606: WD   : Uptime 3 ConnectFailures 0 FreeMem 17768 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
181745: [P150] RF GW: 181729 - 181745 = 16
181746: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
181747: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 0;
181747: [P150] RF GW: 1 - Bytes:10; Counter:1; RSSI:-79 (0xF5); LQI: 0x30
181747: [P150] RF GW: 2 - 00,
181747: [P150] RF GW: 3 - Ignore message, not our networkID.
181885: [P150] RF GW: 181884 - 181885 = 1
181886: [P150] RF GW: rxBYTES: 0D (hex) 13 (dec); getMarcState 0D
181887: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 0;
181887: [P150] RF GW: 1 - Bytes:10; Counter:1; RSSI:-72 (0x04); LQI: 0x32
181887: [P150] RF GW: 2 - 00,
181887: [P150] RF GW: 3 - Ignore message, not our networkID.
186531: [P150] RF GW: 186531 - 186531 = 0
186532: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
186533: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 7;
186533: [P150] RF GW: 1 - Bytes:13; Counter:2; RSSI:-80 (0xF4); LQI: 0x32
186533: [P150] RF GW: 2 - 20,00,52,01,
186533: [P150] RF GW: 3 - Ignore message, not our networkID.
191344: [P150] RF GW: 191333 - 191344 = 11
191345: [P150] RF GW: rxBYTES: 1C (hex) 28 (dec); getMarcState 0D
191346: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
191346: [P150] RF GW: 1 - Bytes:13; Counter:1; RSSI:-78 (0xF7); LQI: 0x31
191346: [P150] RF GW: 2 - 20,00,52,01,
191346: [P150] RF GW: 3 - Ignore message, not our networkID.
192333: [P150] RF GW: 2 - 
192333: [P150] RF GW: 3 - Ignore message, not our networkID.
192434: [P150] RF GW: RX FIFO OVERFLOW -> idle, flush RX FIFO and switch to RX state.
192651: [P150] RF GW: 192648 - 192651 = 3
192652: [P150] RF GW: rxBYTES: 11 (hex) 17 (dec); getMarcState 0D
192653: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:1; DEST:3; ORG.SRC:1; ORG.DEST:3; Ntwrk:0008336f;Type: 7;
192653: [P150] RF GW: 1 - Bytes:14; Counter:2; RSSI:-77 (0xF9); LQI: 0x2E
192653: [P150] RF GW: 2 - 30,00,40,00,05,
192653: [P150] RF GW: 3 - Ignore message, not our networkID.
192671: [P150] RF GW: 192654 - 192671 = 17
192672: [P150] RF GW: rxBYTES: 1F (hex) 31 (dec); getMarcState 0D
192673: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 6;
192673: [P150] RF GW: 1 - Bytes:9; Counter:2; RSSI:-73 (0x02); LQI: 0x2F
192673: [P150] RF GW: 2 - 
192673: [P150] RF GW: 3 - Ignore message, not our networkID.
192991: [P150] RF GW: 192971 - 192991 = 20
192992: [P150] RF GW: rxBYTES: 26 (hex) 38 (dec); getMarcState 0D
192993: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 7;
192993: [P150] RF GW: 1 - Bytes:16; Counter:12; RSSI:-70 (0x09); LQI: 0x33
192993: [P150] RF GW: 2 - 50,00,41,00,05,00,11,
192993: [P150] RF GW: 3 - Ignore message, not our networkID.
193276: [P150] RF GW: 193273 - 193276 = 3
193277: [P150] RF GW: rxBYTES: 26 (hex) 38 (dec); getMarcState 0D
193278: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 7;
193278: [P150] RF GW: 1 - Bytes:16; Counter:12; RSSI:-71 (0x06); LQI: 0x30
193279: [P150] RF GW: 2 - 50,00,41,00,05,00,11,
193279: [P150] RF GW: 3 - Ignore message, not our networkID.
194903: [P150] RF GW: 2 - 
194903: [P150] RF GW: 3 - Ignore message, not our networkID.
195221: [P150] RF GW: 195202 - 195221 = 19
195222: [P150] RF GW: rxBYTES: 33 (hex) 51 (dec); getMarcState 0D
195223: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:1; ORG.SRC:3; ORG.DEST:1; Ntwrk:0008336f;Type: 7;
195223: [P150] RF GW: 1 - Bytes:16; Counter:13; RSSI:-76 (0xFC); LQI: 0x30
[... 171 lines cut ...]
451605: WD   : Uptime 8 ConnectFailures 0 FreeMem 19144 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
460524: [P150] RF GW: Sent command for 'join' to DUCO unit
460531: [P150] RF GW: 0 - SendJoinPacket()
460531: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
460531: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
460531: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
460531: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
460776: [P150] RF GW: 460761 - 460776 = 15
460777: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
460778: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 2;
460778: [P150] RF GW: 1 - Bytes:13; Counter:7; RSSI:-74 (0x01); LQI: 0x2F
460779: [P150] RF GW: 2 - 00,00,7C,3E,
460779: [P150] RF GW: 3 - Received messagetype: JOIN2
460779: [P150] RF GW: 4 - processJoin2Packet()
460779: [P150] RF GW: 5 - joinPacketValidNetworkId()
460779: [P150] RF GW: 6 - SendJoin3Packet: INVALID join2 packet received!
480353: [P150] RF GW: UPDATE by PLUGIN_READ
481605: WD   : Uptime 8 ConnectFailures 0 FreeMem 19168 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
491530: [P150] RF GW: PLUGIN_WEBFORM_LOAD
493589: [P150] RF GW: PLUGIN_WEBFORM_LOAD
511623: WD   : Uptime 9 ConnectFailures 0 FreeMem 19144 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
```
</details>

_Originally posted by @dgokoel in https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/123#issuecomment-1375588910_

arnemauer, 2023-01-27:
@dgokoel Als ik kijk naar je eerste log dan ontvangt de gateway een beter signaal van je bedieningsschakelaar dan van je Ducobox. De bedieningsschakelaar zal dan als een repeater gaan functioneren:
``` 
460524: [P150] RF GW: Sent command for 'join' to DUCO unit
460531: [P150] RF GW: 0 - SendJoinPacket()
460531: [P150] RF GW: 1 - SENT message (outboxQ 0): SRC:1; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:00007c3e;Type: 1;
460531: [P150] RF GW: 2 - Bytes:13; Counter:1; WaitForAck: 0;
460531: [P150] RF GW: 3 - 00,01,00,00,7C,3E,08,02,01,00,00,00,0C,
460531: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1
460776: [P150] RF GW: 460761 - 460776 = 15
460777: [P150] RF GW: rxBYTES: 10 (hex) 16 (dec); getMarcState 0D
460778: [P150] RF GW: 0 - RECEIVED message (inboxQ 0):  SRC:3; DEST:0; ORG.SRC:1; ORG.DEST:0; Ntwrk:0008336f;Type: 2;
460778: [P150] RF GW: 1 - Bytes:13; Counter:7; RSSI:-74 (0x01); LQI: 0x2F
460779: [P150] RF GW: 2 - 00,00,7C,3E,
460779: [P150] RF GW: 3 - Received messagetype: JOIN2
460779: [P150] RF GW: 4 - processJoin2Packet()
460779: [P150] RF GW: 5 - joinPacketValidNetworkId()
460779: [P150] RF GW: 6 - SendJoin3Packet: INVALID join2 packet received!
```

Daar kan de code van de gateway nog niet mee omgaan. Hij zal rechtstreeks met de Ducobox moeten communiceren. Je kan de gateway daarom dichter bij de Ducobox neerzetten.

dgokoel, 2023-03-14:
Hoi @arnemauer, ik ben er weer voor gaan zitten en het is gelukt! Had ook een wifi antenne gekocht om te kijken of ik dan dichterbij de box kon komen en dan toch wifi verbinding kon krijgen, dat was het helaas niet. Maar ik denk dat ik een soort perfecte spot heb gevonden waardoor ik nu wel heb kunnen joinen! Om nu ook de seriële verbinding tegelijk te laten werken heb ik een langere rj11 kabel nodig dan geleverd dus die ga ik bestellen. Hartelijk dan in ieder geval voor al je hulp!