arnemauer/Ducobox-ESPEasy-Plugin #141: Ventilation gateway beschikbaarheid

MelvinvanMelzen, 2023-04-05:
Hoi Arne, 

Ik was benieuwd of de prebuilt gateway weer beschikbaar komt? Ik zie op je webshop dat de (meeste?) onderdelen binnen zijn. Eventueel zou een kit die nog gesoldeerd/monteerd en geflashed moet worden ook wel prima vinden. Thanks! 

Groet Melvin

kaazvaag, 2023-05-15:
Hallo Arne,

Ik zou ook graag een prebuilt Ventilation Gateway bij je bestellen. Enig idee wanneer ze weer beschikbaar zijn? Op de website (https://ventilationgateway.com/) zijn al 5 maanden geen updates m.b.t. beschikbaarheid.

Groeten

mckeehnen, 2023-10-15:
Hallo Arne,

Ik heb dezelfde vraag als bovenstaande. Als compleet niet meer mogelijk is dan wellicht de print?

grt Marco

tonnie84, 2023-11-07:
Hallo Arne, gaaf project!
Ik zou ook graag een prebuilt gateway bij je kopen. 
Vr gr

NigelVanHattum, 2024-02-03:
Ik zie nu op de website dat deze ergens in februari beschikbaar komen. Heb je enige indicatie wanneer exact @arnemauer ? Met de warme zomer op komst lijkt het me erg fijn om wat meer controle over de ventilatie te hebben :-)

arnemauer, 2024-02-11:
Dat was wel het plan maar door drukte gaat mij dat niet lukken. Ik ga kijken of ik in de huidige plugins ondersteuning in kan bouwen voor de esp32. Het volledig ombouwen naar één plugin voor alle data komt dan later.

BrendanxP, 2024-04-02:
Ook hier is interesse in een kit. Ik zie graag een melding bij beschikbaarheid :)

jflieben, 2024-05-30:
idem!

paddenstoeltje, 2024-06-15:
Ik ben ook zeker geïnteresseerd voor mijn duco premium 325 makkelijk aan HA te hangen!

cmseasy, 2024-07-09:
Is de USB kabel op dit moment wel leverbaar? Hier is wat onduidelijkheid over: op de homepage staat dat de kabel ook uitverkocht is, echter ik kan de kabel in de winkelwagen plaatsen en betalen.

Ook ik ben zeer geïnteresseerd in de (prebuilt) Ventilation Gateway voor mijn Ducobox Energy Comfort. 
Is er enig zicht over de leverbaarheid?

arnemauer, 2024-07-09:
De kabels zijn bestelbaar :) Ondertussen ben ik druk bezig met het tunen van de RF antenne voor de nieuwe versie van de gateway. Ik hoop dat ik komende maand daar de onderdelen voor binnenkrijg.

cmseasy, 2024-07-09:
> De kabels zijn bestelbaar :) Ondertussen ben ik druk bezig met het tunen van de RF antenne voor de nieuwe versie van de gateway. Ik hoop dat ik komende maand daar de onderdelen voor binnenkrijg.

Goed om te horen.
Ga ik het kabeltje bestellen. En hou ventilationgateway.com in de gaten voor de gateway.

cmseasy, 2024-07-11:
Kabeltje ontvangen en direct getest op de Duco Energy Comfort D325 met Duco Netwerk Tools versie 3.10.4.0 (oudere versie, zonder cloud). Alle sensoren zijn uit te lezen. Om kabelloos te kunnen werken heb ik een usp/ip applicatie op een nog ergens rondslingerende Raspberry 2 gezet. Werkt ok vanaf zolder naar werkkamer.

Dat was even off-topic maar ter info.
On topic, in afwachting van een connectie naar Homeassistant met de Gateway.

@arnemauer 👍

NigelVanHattum, 2024-11-20:
Hi @arnemauer , 

Mijn zelf gebouwde oplossing is helaas alles behalve stabiel. Zie jij nog de mogelijkheid om deze Gateway beschikbaar te maken?

Yihaaa73, 2025-01-04:
Komt de RF gateway ooit nog beschikbaar?