arnemauer/Ducobox-ESPEasy-Plugin #142: RF gateway blijft disabled

bartsnijder, 2023-07-20:
Ik probeer de CC1101 RF gateway toe te voegen, maar na submitten blijft de plugin op disabled staan.

arnemauer, 2023-07-20:
Je geeft te weinig informatie om je verder te helpen.

- Heb je alle stappen van de wiki doorlopen?
- Wat is het probleem?
- Wat heb je zelf al geprobeerd?

bartsnijder, 2023-07-20:
Ik heb de wiki doorlopen.
Ik gebruik een D1 mini pro met eem RF Transceiver 868MHz CC1101 (Build: [ESP_Easy_mega_20230518_custom_ESP8266_4M1M May 18 2023])
De seriele gateway is operationeel maar de DUCO ventilation remote kan niet enabled worden. Ik volg de exacte stappen van de site, maar na het selecteren van "enable" en submit, blijft de unit disabled.

bartsnijder, 2023-10-30:
Niet meer gereageerd, excuses. De CC1101 module bleek defect, na 1 op 1 vervanging werkt alles zoals verwacht. Bedankt voor het meedenken

closed 2023-10-30 by bartsnijder