arnemauer/Ducobox-ESPEasy-Plugin #149: Getway maak geen verbinding met Duco box

xander450, 2024-01-30:
Hoi ik heb een Duco box energy comfort
En deze wil niet venbinden met mijn rf getway
Krijg wel de signaal te zien van de afstandsbediening in de log maar geen responge van de Duco box

arnemauer, 2024-01-30:
Kun je aangeven welke stappen je exact doorloopt? En kun je een log hier plaatsen?

xander450, 2024-01-30:
Ik heb de Log alewn als Foto en Screenshot
Ik heb alles volgens de rf config gedaan 
Heb de Duco box in paring mode gezet via de inst knop

[image]
[image]

arnemauer, 2024-01-30:
Volgende keer even het knopje copy log to clipboard gebruiken :)

Welke hardware gebruik je?
Kun je een log sturen van het activeren van de plugin?

xander450, 2024-01-30:
Ik gebruik een wmos di 1 mini pro
Ben momenteel niet thuis als ik thuis ben 
Dan stuur ik de log door

xander450, 2024-01-30:
Hoi ik hoop dat het zo duidelijk is

4678: WIFI : Entering processGotIP()
4680: WIFI : DHCP IP: 192.168.1.103 (VENTILATION-GATEWAY) GW: 192.168.1.1 SN: 255.255.255.0 DNS: 192.168.1.1 / (IP unset)   duration:
4682: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
4790: NTP  : NTP replied: delay 61 mSec Accuracy increased by 124 msec
4792: Time set to 1706642619.124
4793: Current Time Zone: STD time start: 2024-10-27 03:00:00 offset: 0 min
4797: Local time: 2024-01-30 19:23:39
4803: WIFI : Arduino wifi status: WL_CONNECTED 3 ESPeasy internal wifi status: IP 
4804: UDP : Start listening on port 8266
4805: firstLoopConnectionsEstablished
4811: WiFi : Missed connected event
4813: WIFI : Connected! AP: JX-Wifi (C0:06:C3:3F:10:9A) Ch: 4 Duration: 1013 ms
4813: WiFi : WiFi services initialized
4814: mDNS : Starting mDNS...
4816: mDNS : Started, with name: VENTILATION-GATEWAY.local
10616: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
10960: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
13174: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
13794: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
14515: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
14707: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
18993: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
19004: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
19107: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
19209: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
19624: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
24218: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
24432: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
25299: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
26274: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
26391: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
26585: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
26687: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
27909: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
28214: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
29348: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
30083: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
30466: SaveToFile: free stack after: 3200
30470: SaveToFile: free stack: 3328
30543: FILE : Saved config.dat offset: 0 size: 1320
30543: SaveToFile: free stack after: 3328
30544: Skip saving SecuritySettings, not changed
30548: mDNS : Starting mDNS...
30549: mDNS : Started, with name: VENTILATION-GATEWAY.local
30550: [P150] RF GW: EXIT PLUGIN_150
30737: [P150] RF GW: PLUGIN_WEBFORM_LOAD
31101: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
31114: static_file: esp.css
31142: HTML : Request file esp.css
31792: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
31803: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
32134: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
32819: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
32823: WD   : Uptime 1 ConnectFailures 0 FreeMem 10112 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
33358: LoopStats: shortestLoop: 65 longestLoop: 903351 avgLoopDuration: 99.57 loopCounterMax: 461538 loopCounterLast: 293498
33358: Scheduler stats: (called/tasks/max_length/idle%) 293492/1872/7/79.60
33766: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
33935: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
34259: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
34931: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
35029: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
35074: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
35276: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
35378: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
36108: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
36303: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
36984: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
37226: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
[... 29 lines cut ...]
53135: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
53397: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
54230: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
54727: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
55370: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
55643: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
56625: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
56868: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
57896: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
58817: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
58869: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
59022: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
59124: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
59906: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
61054: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
61378: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
[... cut ...]
63358: LoopStats: shortestLoop: 65 longestLoop: 903351 avgLoopDuration: 86.96 loopCounterMax: 461538 loopCounterLast: 334854
63358: Scheduler stats: (called/tasks/max_length/idle%) 334853/2035/7/91.10
63426: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
64954: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
65056: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
66386: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
67184: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm

arnemauer, 2024-01-30:
Heb je de plugin op debug staan? En de weblog ook op debug?
Kun je een foto maken van de cc1101 module die je gebruikt?

xander450, 2024-01-31:
Ik heb even schreenshot gemaakt van alles 
[image]
[image]
[image]
[image]

xander450, 2024-01-31:
Ik heb nu net een nieuwe cc11001 gesoldeerd en op een node mcu gezet heb hier het zelfde probleem

arnemauer, 2024-01-31:
> Ik heb nu net een nieuwe cc11001 gesoldeerd en op een node mcu gezet heb hier het zelfde probleem

Kun je een foto posten van je cc1101 met antenne?

xander450, 2024-02-01:
[image]
[image]

arnemauer, 2024-02-11:
Wat is de afstand tussen de ducobox en de module? Je kan de weblog open houden en dan vanaf 4 meter telkens 30cm dichter bij de ducobox staan. Ondertussen met een bedieningsschakelaar de stand veranderen. En dan kijken of je pakketje voorbij ziet komen.

Bram81, 2024-09-11:
Ik had hetzelfde probleem, maar met versie van 18 05 2023 werkte alles direct.