arnemauer/Ducobox-ESPEasy-Plugin #152: Docs: Home Assistant MQTT config

NigelVanHattum, 2024-03-16:
De Wiki pagina voor de connectie met Home Assistant mist een aantal cruciale screenshots. Dit zorgt er nu voor dat ik het nu nog niet volledig kan implementeren. 

Ik maak zelf al wel gebruik van Zigbee2MQTT op mijn homeassistant. Hopelijk is dit geen blocker voor de DUCO gateway. 

Ik mis met name de configuratie van de connectie in de configuration.yaml

arnemauer, 2024-03-16:
Heb je hier iets aan? https://github.com/arnemauer/Ducobox-ESPEasy-Plugin/issues/24

NigelVanHattum, 2024-03-17:
Dank, dit heeft me zeker op weg geholpen. Ik gebruik zelf alleen de RF-Gateway, dus iets meer uitgekleed en een klein beetje aangepast. 
Ik heb voor de RF gateway nu een sensor die de status uitleest en een selector waarmee je de staat kunt kiezen. 

Dit is mijn huidige setup, misschien kun je deze toevoegen aan de wiki? 

Sensor om de huidige staat uit te lezen: 
```yaml
mqtt:
  - sensor:
      - name: "Ventilation mode"
        unique_id: duco_ventilatie_status
        state_topic: "{{Unit Name}}Ventilation_gateway_RF/Ventilationmode"
        value_template: >-
          {% set status = value | int %}
          {% if status == 13 %}
          {{'Perm. Hoog'}}
          {% elif status == 12 %}
          {{'Perm. Middel'}}
          {% elif status == 11 %}
          {{'Perm. Laag'}}
          {% elif status == 3 %}
          {{'Hoog'}}
          {% elif status == 2 %}
          {{'Middel'}}
          {% elif status == 1 %}
          {{'Laag'}}
          {% elif status == 0 %}
          {{'Auto'}}
          {% elif status == 4 %}
          {{'Not Home'}}
          {% else %}
          {{'Undefined'}}
          {% endif %}
```

Selector voor de ventilatie stand:
```yaml
mqtt:
  - select:
      - name: "Ventilation mode select"
        unique_id: duco_ventilatie_select
        retain: true
        options: 
          - 'Perm. Hoog'
          - 'Perm. Middel'
          - 'Perm. Laag'
          - 'Hoog'
          - 'Middel'
          - 'Laag'
          - 'Auto'
          - 'Not Home'
        state_topic: "Smart-DUCO/Ventilation_gateway_RF/Ventilationmode"
        value_template: >-
          {% set status = value | int %}
          {% if status == 13 %}
          {{'Perm. Hoog'}}
          {% elif status == 12 %}
          {{'Perm. Middel'}}
          {% elif status == 11 %}
          {{'Perm. Laag'}}
          {% elif status == 3 %}
          {{'Hoog'}}
          {% elif status == 2 %}
          {{'Middel'}}
          {% elif status == 1 %}
          {{'Laag'}}
          {% elif status == 0 %}
          {{'Auto'}}
          {% elif status == 4 %}
          {{'Not Home'}}
          {% else %}
          {{'Undefined'}}
          {% endif %}
        command_topic: "Smart-DUCO/ESP1/cmd"
        command_template: >-
          {% if value == 'Perm. Hoog' %}
          {{'VENTMODE,PERMANENTHIGH'}}
          {% elif value == 'Perm. Middel' %}
          {{'VENTMODE,PERMANENTMIDDLE'}}
          {% elif value == 'Perm. Laag' %}
          {{'VENTMODE,PERMANENTLOW'}}
          {% elif value == 'Hoog' %}
          {{'VENTMODE,HIGH'}}
          {% elif value == 'Middel' %}
          {{'VENTMODE,MIDDLE'}}
          {% elif value == 'Laag' %}
          {{'VENTMODE,LOW'}}
          {% elif value == 'Auto' %}
          {{'VENTMODE,AUTO'}}
          {% elif value == 'Not Home' %}
          {{'VENTMODE,NOTHOME'}}
          {% else %}
          {{'VENTMODE,AUTO'}}
          {% endif %}

```

closed 2024-12-22 by NigelVanHattum