arnemauer/Ducobox-ESPEasy-Plugin #158: DIY RF Gateway kan niet verbinden met Duco Comfort 325

EwoutBergsma, 2025-01-05:
Dag Arne,

Bedankt voor dit prachtige project. Ik heb de wiki gelezen, maar verbinden met mijn Duco lukt niet. Hier wat details:
- De Gateway hangt op ongeveer een meter afstand van de Duco Comfort 325
- Ik draai de 20230518 firmware
- Ik gebruik zo'n standaard CC1101 transceiver module

Als ik de Duco in installer mode zet, en vervolgens op "join" klik in de browser krijg ik de volgende log:

426506: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
426889: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
427300: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
427514: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
427638: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
428221: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
428423: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
428655: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
429150: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
429802: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
430065: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
430576: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
430679: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
431056: WiFi : Set TX power to 11dBm sensitivity: -72dBm RSSI: -63dBm
431187: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
431399: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
431500: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
433150: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
433315: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
433446: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
433548: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
434330: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
434496: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
434668: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
434780: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
435513: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
435618: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
435901: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
436003: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
436097: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
436260: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
436271: static_file: esp.css
436272: Serve 304: 3528815976 esp.css
436272: HTML : Request file esp.css
436417: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
436618: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
436644: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
436925: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
437027: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
437502: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
437846: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
438048: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
438746: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
438977: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
439278: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
439591: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
439863: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
440096: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
440136: [P150] RF GW: UPDATE by PLUGIN_READ
440998: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
441229: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
441485: WD   : Uptime 7 ConnectFailures 0 FreeMem 10400 WiFiStatus WL_CONNECTED 3 ESPeasy internal wifi status: Conn. IP Init
441964: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
442348: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
442373: LoopStats: shortestLoop: 100 longestLoop: 5447778 avgLoopDuration: 133.77 loopCounterMax: 300000 loopCounterLast: 219421
442374: Scheduler stats: (called/tasks/max_length/idle%) 1429733/13336/7/89.05
442671: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
442863: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
443095: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
443219: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
443380: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
[... 139 lines cut ...]
486111: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
486279: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
486482: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
487245: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
487357: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
487609: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
487711: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
488333: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
488492: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
488945: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
489248: WiFi : Set TX power to 7dBm sensitivity: -72dBm RSSI: -59dBm
489502: WiFi : Set TX power to 10dBm sensitivity: -72dBm RSSI: -62dBm
489767: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
490278: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
490310: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
490381: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
490745: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm
491398: WiFi : Set TX power to 8dBm sensitivity: -72dBm RSSI: -60dBm
491562: WiFi : Set TX power to 9dBm sensitivity: -72dBm RSSI: -61dBm

Twee dingen vallen mij op:
- de log wordt gespamt door TX power van de WIFI, dit had ik niet bij oudere versies van de firmware?
- er lijkt geen reactie komen op "468189: [P150] RF GW: 4 - Joinpacket sent. DucoDeviceState = ducoDeviceState_join1"?

Alvast bedankt!

EwoutBergsma, 2025-03-13:
Hmm.. interessant. Ik ga proberen hier binnenkort naar te kijken. 

Wat me wel opvalt is dat je in #160 zegt/impliceert(?) dat een wifi antenna beter werkt dan de originele antenna, want een wifi antenna is gebouwd voor 2.4ghz, en de RF-gateway werkt op 868Mhz? Of zie ik dit verkeerd?

EwoutBergsma, 2025-04-01:
Ik heb een poging gedaan, zonder succes. Wat ik heb gedaan:
- RF gateway dichterbij geplaatst (óp de Duco, en een meter afstand)
- de log aangezet en geopend 
- de Duco bediend met de bijgeleverde draadloze afstandsbediening
- gezocht naar een melding omtrent de communicatie, zonder resultaat

Haapster, 2025-04-01:
Dat klinkt alsof je hetzelfde hebt gedaan als ik. Vervelend dat je geen communicatie ziet verder, dat gaf bij mij de doorslag om verder te blijven prutsen. Ik vrees dat ik je niet verder kan helpen hiermee, helaas zijn mijn technische skills niet zo denderend...

arnemauer, 2025-04-10:
Kun je een log posten met daarin het moment dat je de plugin activeert?

RabbitMD, 2025-04-11:
Het lijkt op een zelfde issue dat ik heb / had, ik heb 325 unit, met de laatste versie van de bin werkt de join niet wat ik ook probeer qua afstand / settings.
pak ik binary 20200608 Firmware aa380ce 08-06-2020 (ESPEasy mega-20200515).bin , dan werkt de Join bij mij direct, ook op langere afstanden. 
Mogelijk iets met firmware versies van de 325 units?