arnemauer/Ducobox-ESPEasy-Plugin #161: Join knop wordt niet zichtbaar

tvesch, 2025-03-08:
Hallo, wanneer ik de stappen doorloop om de rf gateway in te stellen loopt het vast bij de stap die zegt op "Join" te drukken. Heb meerdere firmwares geprobeerd, maar bij alle loop ik tegen hetzelfde probleem aan. Enig idee of em hoe dit op te lossen is? Kan dit aan het de hardware of antenne zelf liggen?

LRAB1, 2025-06-20:
Kijk eens of je TX lijn een weerstandje heeft aan de CH340 kant, ik heb het vermoeden dat de verbinding met de CC1100 niet op gang komt als het weerstandje op die lijn ontbreekt.

Bron: ik had hetzelfde probleem, ben alleen een andere weg ingeslagen dan RF gateway in te zetten.

19Marco88, 2025-08-21:
[image]
Dit geeft hij aan, ik denk dat de CC1101 niet goed werkt.
Maar als de soldeerwerk is gewoon goed, en ik heb ook de juiste type.

Hoe kan ik eventueel de CC1101 configureren? Ik heb de stappen gevolgd zoals ze hier vermeld staan.

19Marco88, 2025-08-21:
Ik heb de draden gecheckt en dat ziet er goed uit.
Alleen heb nu niet de mogelijkheid om ze door te meten, dat kan ik pas over 1,5 week weer doen.
Zou het eventueel ook aan de CC1101 kunnen liggen dat die niet goed is? Ik zal nog even checken of ik hem goed heb aangesloten.