arnemauer/Ducobox-ESPEasy-Plugin #164: Repo abandoned?

LRAB1, 2025-06-12:
Should this repo not be closed? It seems to no longer be maintained or updated.

arnemauer, 2025-06-18:
I'm almost done with a new hardware version but i'm building a new house. I hope to have more time by the end of the year to finish it.

closed 2025-06-18 by arnemauer

LRAB1, 2025-06-18:
Nice! I'm currently working on an integration for Homey based on my workarounds. Perhaps I could integrate your new solution when it is ready. Seeing as the HP23 does not have 868mhz support, perhaps it'd be nice to bridge through your gateway.

DionVermeulen, 2025-12-12:
> I'm almost done with a new hardware version but i'm building a new house. I hope to have more time by the end of the year to finish it.

Hi,

Any update on this? I'm currently renovating my house and implenting Duco Focus Box with zones. It would be really nice if the new project is (almost) fnished :).